FreeCAD/FreeCAD #18891: When object is selected in sidebar it is not highlighted in graphic area

hramrach, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When an object is selected by clicking the graphic area it is highlighted.

However, when object is selected in sidebar the object is not highlighted. Instead some cube is drawn in the graphic area.

### Full version info

```shell
OS: openSUSE Leap 15.6 (xmonad/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.6.15, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.7.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Anything else?

Seems to be regression since 0.20

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hramrach, 2025-01-05:
No, I don't remember it correctly.

FreeCAD has some 'prelight' feature which highlights an object by hovering over it.

In 1.0.0 when hovering over an area one specific object from that area is always highlighted, and selected on click.

However, on 0.20.2 hovering over an area with multiple objects (eg. lines that overlap partially) randomly highlights different objects making it possible to make out different objects due to changes in highlighting, and select specific objects.

hramrach, 2025-01-05:
Additional reference https://xkcd.com/1172/

maxwxyz, 2025-01-06:
Duplicate https://github.com/FreeCAD/FreeCAD/issues/9205 and https://github.com/FreeCAD/FreeCAD/issues/11215

closed 2025-01-06 by maxwxyz