FreeCAD/FreeCAD #18894: 4	Issue "Create external projection geometry" worked perfectly in 1.0.0.39109 (Git) Conda but does not work in 1.1.0dev.39750 (Git) Conda

AWeighell, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to create issue. (Tested also in safe mode).
1.	Part Design WB.  New Part, New Body, 
2.	New Sketch.  Draw line at random angle to ordinate axes. Close sketch.
3.	Create Datum Plane on line.  Rotate plane at random angle such that intersection with body planes is not on ordinate axes.
4.	Create Datum Line (intersection) using Datum Plane and one of body ordinate plates
5.	Select Datum plane.  Create Sk
[TEST PLANES.zip](https://github.com/user-attachments/files/18317230/TEST.PLANES.zip)
etch.  Create external projection geometry by selecting the Datum Line.  Step 5 works perfectly in build 39109 (File Test Planes.FCStd).  
Issue The intersection Datum line is not selectable using in 39750 Create external projection geometry. (File Test Planes dev.FCStd) 

### Full version info

```shell
Issue on build 39750
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.34
  * frame 0.1.1
  * lattice2 1.0.0
  * Render 2024.12.15
  * sheetmetal 0.6.13
  * Silk 0.1.5
  * Trails 2022.1.0
__________________________________
No issue on version 39109
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.34
  * frame 0.1.1
  * lattice2 1.0.0
  * Render 2024.12.15
  * sheetmetal 0.6.13
  * Silk 0.1.5
  * Trails 2022.1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-06:
Cannot reproduce, it works as expected in 
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * parts_library
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5

```

AWeighell, 2025-01-06:
Many thanks for your prompt reply.  I am stuck.  Hopefully, the feature will return at some time on my hardware/OS. I will try it on a Linux install.  I can project the Datum Line into Sketch001 using "Create External Geometry" using 39109 but not 39750. (or the previous dev).  This applies both when the files originated in 39109 and when created from scratch in 39750  
I am guessing that it is something to do with the message when older FC files were opened with the dev editions.  i.e.  
"This file was created with an older version of FreeCAD. Origin axes had incorrect 
placements, which have now been corrected. 
However, if you save this file in the current version and reopen it in an older 
version of FreeCAD, the origin axes will be misaligned. Additionally, if your file 
references these origin axes, your file will likely be broken".

AWeighell, 2025-01-06:
I have repeated the above in two Linux builds.  The outcome is exactly the same.  Projection of a datum line into a sketch works with perfectly with 1.0.0.38641 +468 (Git) Snap 1202 but not with current dev.

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb) 
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * BIM021
  * dodo 1.0.1
  * fasteners 0.5.33
  * Design456
  * Design456.backup1733688069.59679 (Disabled)
  * Design456.backup1733688093.138092 (Disabled)
  * Curves 0.6.51
  * CadbaseLibrary 1.0.3
  * GDT
  * frame 0.1.1

OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/Dark theme/Qt default

pierreporte, 2025-01-10:
The issue also exists with intersections.

I don’t remember how it used to work. Did it generate infinite geometry?

VideoBSO, 2025-01-23:
I stumbled on this issue, trying the [pozidrive tutorial](https://www.youtube.com/watch?v=vOw9g35pr3M) in FC rev `39896 (Git)` on windows.

I found another reproduction example:
- create sketch on `XY` plane
- draw circle constraint to origin and size 50mm
- pad with 10mm
- select `face3`
- add datum point
- create sketch on `YZ` plane
- use external geometry tool -> you can't select the datum point

pierreporte, 2025-01-24:
@maxwxyz Yes, it is needed.

maxwxyz, 2025-01-24:
What kind of geometry is added to a sketch when projecting a datum plane or line, infinite length would not be possible?

pierreporte, 2025-01-24:
In Catia it creates an infinite line. I would expect the same. It could be line a bit longer that the bounding box of the sketch geometry or 3D model currently in the view, that grows automatically if geometry goes beyond. But it should be as if the line is infinite.

jffmichi, 2025-01-24:
Would be nice if the line is infinite but if it's not, e.g. point-on-line constraint still works outside of the line endpoints so still working as expected. It the sketchers doesn't currently support infinite geometry having a finite line would still be better than it not working at all.

Shkolik, 2025-02-20:
It's not possible with current implementation, because here we check for sub element name, but datums doesn't have any:
https://github.com/FreeCAD/FreeCAD/blob/e1512dc0098c3ffa6aa4b42f07d7cedb99899353/src/Mod/Sketcher/Gui/DrawSketchHandlerExternal.h#L100

More to that - we can only project objects that has a Shape, but this new datums being derived from App::whatever doesn't have shape - they are pure geometry/visual objects. 

If we want to be able to project/intersect em we have 2 choices:
1) give em a shape by deriving from Part::Feature 
2) give em a shape by constructing it on a fly - just how it's done for App::Plane projection for example

Do not like 1st variant... will try to play with second. If anyone have any suggestions - let me know.

PS: Digging FreeCAD codebase is kinda fun. Hope I can help with resolving some issues in future :)

commit f3643af8 (2025-03-02): Merge pull request #19747 from Shkolik/sketcher_datums_projection

fix #18894 Sketcher: Create external projection geometry not working with datum objects
closed 2025-03-02 by chennes