FreeCAD/FreeCAD #18895: Consolidate Mini Cube and Main Cube menus

dzid26, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Existence of a menu under mini cube button is not obvious - to me. 

Simplify the UX and make options from the mini cube available under the right-click of the main cube. 

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * CubeMenu
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * kicadStepUpMod 11.4.4
  * Launcher
  * ModernUI (Disabled)
  * Ondsel-defaults.cfg
  * OpenTheme 2024.9.1
  * PieMenu 1.10.1
  * POV-Ray-Rendering
  * ToolbarStyle
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dzid26, 2025-01-06:
Also, Link Actions sub-menu doesn't belong here.

Roy-043, 2025-01-06:
Note that there is no 'main cube menu'. That is just the context menu of the 3D view.

dzid26, 2025-01-06:
@Roy-043 
I would keep those options at the cube. But then I have heard that some people have the cube hidden somehow...

I would:
[image]
[image]

and if nothing is selected:
[image]

I would at least replace with:
[image]

pierreporte, 2025-01-06:
The latest proposal is the following. It would be opened with the small cube menu, that could be changed into a new icon (I advocate for a hamburger style). The context menu could be either dropped completely or replaced with the exact tame menu.

```
Align to selection
Align to sketch plane (Sketcher_ViewSketch)
Isometric
Dimetric
Trimetric
----------------------
Named views          > | Front
Store working view     | Top
Recall working view    | Right
---------------------- | etc.
Fit all
Fit selection
Box zoom
Zoom in
Zoom out
----------------------
(•) Orthographic view
( ) Perspective view
----------------------
Draw style           >
----------------------
Options              > | [ ] Movable navigation cube
                       | ---------------------------
                       | ( ) Top left
                       | (•) Top right
                       | ( ) Bottom left
                       | ( ) Bottom right
```

dzid26, 2025-01-06:
[image]
Cool, hamburger is more meaningful to me than mini-cube. 

> The context menu could be either dropped completely or replaced with the exact tame menu.

But don't drop the Right-click - it is quicker to right-click on big cube with the mouse.

obelisk79, 2025-01-06:
There is no context sensitive right click menu for the cube. What is there to preserve? Or am I just getting list in literary semantics? So what you propose is not just a menu modification, but also a new menu which potentially greatly expands the scope of work required. I feel like it's normal to right click things looking for extra options. 

So we have to ask...

Do we want 
A. a context and dedicated menu (same options)
B. Only a dedicated menu (hamburger)
C. Only a context menu (right click cube widget)
D. Independent context and dedicated menus (different options)

I prefer option C, but it may suffer from a discoverability issue.

pierreporte, 2025-01-06:
The navigation cube does have a context menu, but only if the 3D view is not empty apparently (this looks like an issue).

Option D feels strange already. When right-clicking something, you expect seeing options related to the clicked zone. I am personally in favor of option A, provided that the context menu is independent on the content of the 3D view, and option B if it’s too complicated. Option C indeed lacks discoverability even though it’s functionally sufficient.

dzid26, 2025-01-06:
-> A.

Personally, even C would be ok with me - as a new/returning user I was right-clicking the main cube and discovering the options - to the point I that I thought the cube had independant context :) 
But the dedicated button wouldn't hurt either. Maybe it could be appreciated by people using a stylus/touch in the field, etc.