FreeCAD/FreeCAD #18900: Replace feature eye icon with a checkbox in the tree

dzid26, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The "Visibility" option does so much more than changing the display of a feature. It is a core function of PartDesign and controlling features dependency!

### My proposal:

First - remove checkboxes:
[image]
Selected items are already highlighted.  Checkboxes are redundant. 

Use _checkbox_ icon instead of _eye_ icon for features and groups.
[image]
Keep eye icon for other things, like sketch, shapebinder, origin, etc.

Rename context menu items for the features and groups from:
- Toggle visibility
- Show selection
- Hide selection
to:
- Toggle active
- Activate
- Deactivate

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * CubeMenu
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * kicadStepUpMod 11.4.4
  * Launcher
  * ModernUI (Disabled)
  * Ondsel-defaults.cfg
  * OpenTheme 2024.9.1
  * PieMenu 1.10.1
  * POV-Ray-Rendering
  * ToolbarStyle
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2025-01-06:
This does not seem to be good idea. Checkbox here has dual function - it shows selection state (which is also shown via color) but also allows multiple selection via mouse only. 

In addition, in every kind of item view (be it list view, tree view or any other) checkbox in row often corresponds to selection which is a well known and established pattern that should not be broken. This is used for example in windows file explorer:
[image]

On the other hand, the eye icon is pretty common for denoting visibility of object - it is standard icon for basically all graphics suites:
[image]

dzid26, 2025-01-06:
> On the other hand, the eye icon is pretty common for denoting visibility of object - it is standard icon for basically all graphics suites:

Which is exactly the problem because the eye does not only control the visibility but it's a functional switch and essential to the workflow. 

I believe it's important to separate visibility (sketch, binder) and enabling (features, and groups apparently).

>  selection which is a well known and established pattern that should not be broken

There is many softwares were checkboxes act more as a radio button and a only single one can be selected. 

BUT I would not mind, to use e.g. the _more appropriate_ radio button to be honest and leave the checkboxes alone or let them be optional like in Windows Explorer.

kadet1090, 2025-01-06:
> I believe it's important to separate visibility (sketch, binder) and enabling (features, and groups apparently).

Here we absolutely agree, that is a big issue.  Checkboxes for toggling the visibility are not good, but eye icon for enabling/disabling is also not really great. Can we reframe this issue to target that problem? 

> There is many softwares were checkboxes act more as a radio button and a only single one can be selected.
> 
> BUT I would not mind, to use e.g. the _more appropriate_ radio button to be honest and leave the checkboxes alone or let them be optional like in Windows Explorer.

FreeCAD allows multiple selection and radio buttons imply singular selection. The checkboxes are also disabled by default IIRC, just like windows so I don't really understand? 

[image]

dzid26, 2025-01-06:
Thank you for reminding me about ability to hide the selection checkboxes.  

As for the main issue, I think radio buttons could work. 
But I am not sure if they wouldn't look dated. 

And how would they play visually with the activation of a body - which currently is via context menu. 
[image]

I want to say: the body should have an eye icon _and_ radio button. But I don't feel strong about it.

NomAnor, 2025-01-09:
> Which is exactly the problem because in Freecad the eye does not only control the visibility but it's a functional switch and activating the right feature can be critical.

What do you mean by that? Currently the icon toggles the visibility of features, it does not change how the document objects behave (in contrast to the tip).

dzid26, 2025-01-10:
In a way I agree - it "just changes visibility" of the "features". 

But most users are humans and they live in physical world. Naturally they think features are additive. When they think "hiding a feature", they think "hiding that little fillet on the edge". They don't think that the **whole body** is being replaced with another one that doesn't contain that exemplary fillet. But that's what is happening internally.
One way to make the unaware realize this, would be not to use confusing Visibility nomenclature for features. 

If topo naming problem was solved perfectly, then I guess it wouldn't matter _as much_. 

Also, the user may not realize that selecting faces from old features in the tree, they can help themselves with the topo problems. E.g. when attaching a sketch to a face. New user may not expect so much power from just changing the visibility. 

Also I think the tip should probably follow the active feature by default.

dzid26, 2025-01-10:
Additionally, optionally, I would **not allow** to have no active feature.  - If one wants to hide a body, they can hide it explicitly. 

Pressing the space on a feature could either hide the body or activate previous feature. TBD

NomAnor, 2025-01-10:
> They don't think that the **whole body** is being replaced with another one that doesn't contain that exemplary fillet. But that's what is happening internally.

It's not really, its visual representation changes, but not the shape that get's exported, that is defined by the tip. If you want to remove a fillet, but not delete the feature you can suppress it.

I'm thinking about the timeline in Fusion360 (I used it a long time ago, not sure how correct this is). There you can move a cursor through the features to see past features. This also changes the visual representation of a body but does not "replace" it per se. It could be that the cursor position is always the final shape, that would be your idea to essentially remove the visibility functionallity and merge it with the tip.

This is how I treat the visibility on features and work with PartDesign: The tip is where new features are created and what gets exported. The currently visible feature is the state of the Body at that point.

> As for the main issue, I think radio buttons could work. But I am not sure if they wouldn't look dated.

That would be similar to the Fuion360 cursor.

> Additionally, optionally, I would **not allow** to have no active feature. - If one wants to hide a body, they can hide it explicitly.

That makes sense. I sometimes wonder why the Body is not visible (I have no visible feature).

Maybe whats needed is better documentation of the design/usage ideas behind PartDesign (and other workbenches)? Every CAD program has a different design idea behind them and that needs some learning.

A PartDesign Body is a linear list/container of features and support objects (sketch, datum). The tip defines the feature that is used for actions (new feature, export). Whereas the currently visible feature is just to be able to view past state in the list of steps. New features can be based (e.g. sketch on plane) on older features, which means you can select a plane from a visible feature in the past and create a new sketch at the tip based on it.

Some thoughts that come to mind and might be what you meant:

* The position in the body's feature list could be better visualized (radio button behaviour instead of visibility icon)
* Better differentiation between the "edit state" and the "final state". Export, FEM, CAM, Links should always use the tip feature's shape as the "final state". Whereas the Body represents the "edit state" and shows the shape of the feature at the selected position.

I think the differentiation between both states is an advantage to only having the final state (tip). It needs to be explained to the user (documentation and UI).

dzid26, 2025-01-10:
> It's not really, its visual representation changes, but not the shape that get's exported, that is defined by the tip. If you want to remove a fillet, but not delete the feature you can suppress it.

It looks like that, but it's entirely different object that is presented to you and available for selection. You can see that even the smallest feature can change all face ids that you select. The reason e.g. fillet suppression is even possible is that a topo naming algo finds and reassigns face ids to an object _before_ the operation - on-the-fly. 

So imo, radio buttons wouldn't be just for visuals - it would hopefully help the user understand they are operating on different object and that selecting faces from one feature vs another has dependency consequences. 

Btw, if the above is new to you - I am curious if radio buttons would help you realize this. Lol

> That would be similar to the Fuion360 history cursor.

Yes, a bit similar to Fusion's history. 
But unlike Fusion360, in  Freecad feature selection and tip position are detached and allows for some funny operations like padding a face from before fillet:
[image]

But I think the default behavior should be more like in Fusion - by default, when previous feature is selected the tip should follow the active feature. After doing the operation with the tip moved higher, the features downstream will probably fail but that's ok. Automating tip movement needs to be tackled separately from this issue because as you mentioned the tip also works with external links and exports and we wouldn't want to make that a worse experience.

I do like Freecad flexibility because it teaches you what is actually going on with the underlying geometries. And I like the tree approach where features are together with everything else and not on history bar.