FreeCAD/FreeCAD #18902: [Dark theme] Impossible to tell which context menu items are disabled

dapetcu21, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Disabled context menu items in the dark theme are impossible to distinguish from active items. They don't have greyed out labels (only icons). The only way to tell them apart is that they are not selectable when hovering with the cursor over them.

[image]

### Full version info

```shell
OS: macOS 15.0.1
Architecture: arm64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-06:
If this is (sort of) what you are looking for:

[image]

then just add:

```
QMenuBar:disabled,
QMenuBar::item:disabled,
Qmenu:disabled,
QMenu::item:disabled {
    color: #6b7079;
    background-color: transparent;
}
```

to your _FreeCAD Dark.qss_ file at around line 628 and you can adjust the `#6b7079` to whatever color you think suitable.

dapetcu21, 2025-01-07:
Thank you for the workaround, but if this is indeed the correct solution and doesn't cause issues elsewhere, it should be merged into FreeCAD. 

I believe this is a pretty big usability issue and should be fixed for all users.

MisterMakerNL, 2025-01-07:
Could find! Fix is on the way.

closed 2025-01-27 by yorikvanhavre