FreeCAD/FreeCAD #18906: CAM jobs don't respect visible

sfmissionmark, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When hiding and showing a CAM job the Stock Operations and Model Body do not also hide. 

### Full version info

```shell
OS: macOS 15.1.1
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * TurningAddon 0.3.0
  * woodworking 0.21.2.33771.0.23.1
  * ButlerCues.zip
  * Gridfinity 0.6.0
  * A2plus 0.4.68
  * fasteners 0.5.33
  * ButlerCues
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * ExtremeProDark 2.9.0
  * Curves 0.6.53
```

### Subproject(s) affected?

CAM

### Anything else?

Steps to reproduce:

Create a new document
Add a cube using part design or part
Add new cam job
Increase stock size by some amount.
Add new profile path to create a tool path
Click on cam job in model and hit space bar to hide

Expectation:
Children of the cam job hide.

Result:
Cam job appears to hide
Many paths tools and stock objects do not hide.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

phaseloop, 2025-03-18:
Confirmed issue at 1.1.0dev

tarman3, 2025-03-18:
I solved this issue by macro
https://forum.freecad.org/viewtopic.php?p=809400#p809400

[image]