FreeCAD/FreeCAD #18908: Feature Request: Retain constraints when adding chamfer/fillet

gryphonB-2, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the RealThunder versions of FC there was an ability in a Sketch to chamfer a corner and still retain existing constraints on the wires. This was done by adding a construction Point at the old corner and transferring the constraints to that Point. I used this feature almost universally. 
The included image & file demonstrate this.
One of the major issues FC continues to face is ease of use: the abilities are there & are robust but getting all the ducks in a line can be very painful. 
Case in point: the recent desirable changes to Sketcher: to add a constraint press C, to add a dimension press D and let the context determine what you want. 

In the attached file I have first added 4 construction points, then created the wires I want, added the chamfer then adjusted the constraints. RealThunder's version just auto did this and it just helps to make the whole process more transparent, more 'just do what I want without breaking what I did'  

The attached example file is a valid FCStd file, renamed with '.zip' for upload. 
[Uploading BikeRack.FCStd.zip…]()
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * DynamicData 2.70.0
  * slic3r-tools
```

### Subproject(s) affected?

None

### Anything else?

RealThunder:
[](https://github.com/realthunder/FreeCAD/releases)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gryphonB-2, 2025-01-06:
RealThunder:
[https://github.com/realthunder/FreeCAD/releases](https://github.com/realthunder/FreeCAD/releases)

gryphonB-2, 2025-01-06:
Here is the final Sketch:
[image]

maxwxyz, 2025-01-08:
@PaddleStroke FYI