FreeCAD/FreeCAD #18909: Navigation: Camera position shift glitch when starting viewport rotation

BartekMoniewski, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When starting rotation action, viewport randomly shifts as it was panned by a small distance. This happen in most workbenches and in orthographic view as well as in perspective. In perspective view, with more complex part this glitch can be quite severe, it's very distracting.
Problem occurs when **Navigation > Rotation Mode** is set to **Object Center** or **Drag at Cursor**. I have not encountered it in **Window Center** mode.

Look at attached video to see those shifts.

https://github.com/user-attachments/assets/a73e7a00-66ff-4665-8849-a8ceed504eb1

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.34
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BartekMoniewski, 2025-01-06:
Tested in on latest dev release and behavior is identical.

maxwxyz, 2025-01-08:
@Rexbas FYI

Rexbas, 2025-01-08:
Oh yes I have seen this before and I can confirm this problem, I will have a look at this soon.

PR #18992 (2025-01-19): Gui: Fix camera position shift when starting rotation
Fixes #18909.

Could be back ported.
closed 2025-01-19 by chennes