FreeCAD/FreeCAD #18912: Unicode in the "Open file" dialog

real-artswan, 2025-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
Here we are. Unicode folder names are all off.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Curves 0.6.53
  * InventorLoader 1.5.1
  * sheetmetal 0.6.13
  * 3D_Printing_Tools
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-07:
I think there maybe more to this than simply Unicode as the French accent in `École` shows up fine using the exact same AppImage so it maybe a Cyrillic specific issue:

[image]

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.3
  * sheetmetal 0.6.13
  * Behave-Dark-Colors 0.1.1
  * Alternate_OpenSCAD 1.0.0
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.34
  * Curves 0.6.53
  * Assembly4 0.50.15
  * A2plus 0.4.68

```

xtemp09, 2025-01-07:
@Syres916, 

[image]

----

@real-artswan, please, upload your locale output.

xtemp09, 2025-01-07:
The fonts used have characters missing or system paths don't contain appropriate fonts. @real-artswan, what are your _System Settings_ → _Appearance_ → _Fonts_ settings?

real-artswan, 2025-01-07:
[image]
I use KDE 
[image]

Have never noticed such behaviour in other applications, It seems it is FreeCAD only.

real-artswan, 2025-01-07:
FS is default ext4 but I'm not sure if that panel displays real folder names, it looks like some predefined list. May it read it from some config file or API call? Dolphin and mc and terminal and everything else display all sorts of unicode without any issues (French, Japanese, Chinese, etc).

[image]

BTW all of that cryptic items including `frame` (have no idea what it should be and where it is getting it) are pointed to the root `/`. The `apps` one I added in the Dolphin as a custom "Places" item, `DEVELOPMENT` is added the same way but it's not displayed in the FreeCAD dialog.
[image]

real-artswan, 2025-01-07:
I'm not familiar with Qt but I think it's related to so called "Sidebar URLs"

https://runebook.dev/en/articles/qt/qfiledialog/setSidebarUrls

real-artswan, 2025-01-07:
https://github.com/FreeCAD/FreeCAD/blob/278ce803907ef72548d7ac761613038ac435c481/src/Gui/FileDialog.cpp#L97

maxwxyz, 2025-01-07:
is this also happening with the weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

real-artswan, 2025-01-07:
[image]

FYI

xtemp09, 2025-01-08:
@real-artswan, _LMMS_ contains the same code.[^1] Could you launch _LMMS_ and see if it contains the same bug?[^2]

[^1]: https://github.com/LMMS/lmms/blob/master/src/gui/modals/FileDialog.cpp#L39
[^2]: https://github.com/LMMS/lmms/releases

real-artswan, 2025-01-08:
@xtemp09 
Actually It contains correct ukrainian folders, so the problem is only with particular items.
[image]

xtemp09, 2025-01-08:
Since the problem is present in other _Qt_ software, I conclude, that the bug originates in _Qt_.

maxwxyz, 2025-01-08:
Would this still be an issue with Qt6?

real-artswan, 2025-01-08:
Here is some suspicious encoding conversion, LMMS has the same

https://github.com/FreeCAD/FreeCAD/blob/278ce803907ef72548d7ac761613038ac435c481/src/Gui/FileDialog.cpp#L421

real-artswan, 2025-01-08:
I fixed it by removing `~/.config/FreeCAD/user.cfg`
[image]

maxwxyz, 2025-01-08:
So could this issue be closed then?
It would be great if we knew what entry in the user.cfg broke this.

xtemp09, 2025-01-08:
@real-artswan, can you restore the deleted config from Trash and upload it here?

real-artswan, 2025-01-08:
Yes I think it can be closed, I was wondering what's wrong with it too but couldn't find anything special there. 
here is a backup
[user.cfg.broken-backup.txt](https://github.com/user-attachments/files/18352380/user.cfg.broken-backup.txt)

xtemp09, 2025-01-09:
@real-artswan, could you also upload the list of your environment variables?

real-artswan, 2025-01-09:
[my-envs.txt](https://github.com/user-attachments/files/18361013/my-envs.txt)

closed 2025-01-09 by maxwxyz