FreeCAD/FreeCAD #18915: Version 1.0.0: slowness, error messages, incompatible with 0.21 projects

BrunoMeunier, 2025-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Once V1.0 is installed to take advantage of the improvements compared to V0.21, it is impossible to work with FreeCAD: a multitude of error messages appear indicating inappropriate constraints that were not reported in 0.21. All recalculations are very slow, or even freeze FreeCAD. Boolean operations no longer work.
Worse: version 0.21 still installed on my computer has inherited the defects of version 1.0.0, and in particular in terms of error messages! As a result, it is now impossible for me to work with FreeCAD 0.21 and 1.0.
Extremely disappointed! And very angry because I will have to change mechanical CAD.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * Assembly4 0.50.8
  * dxf-library
  * sheetmetal 0.4.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-01-07:
Please try to restart in safe mode https://wiki.freecad.org/Std_RestartInSafeMode
then share a failing file.

dzid26, 2025-01-08:
Very similar symptoms as https://github.com/FreeCAD/FreeCAD/issues/18205

FEA-eng, 2025-09-27:
@maxwxyz Should it be closed due to the lack of specific information? Unless it's worth investigating e.g. the performance part mentioned also [here](https://github.com/FreeCAD/FreeCAD/issues/18205#issuecomment-2530042212).