FreeCAD/FreeCAD #18917: BIM: Arch_Space not possible with boundary elements

maxwxyz, 2025-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am not able to create a new Arch_Space in the BIM WB by selecting boundary objects or faces.
I can create a space from a single solid by selecting it, also e.g. with a wall or slab object, but to create a space for a room I cannot create it by selecting all boundary wall faces, even when selecting a limiting face on the top and bottom:
[image]

According to the wiki it should be possible to create a space by selecting boundary faces too.

[Boundary_Space.zip](https://github.com/user-attachments/files/18331306/Boundary_Space.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-07:
@semhustej are you able to reproduce this?

yorikvanhavre, 2025-01-07:
The problem in your example is probably that the walls are crossing. Building spaces from boundaries is still very tricky and error-prone (see also #6091).

We need a better algorithm here...

semhustej, 2025-01-07:
Additionally I am not able to add space boundary using the Arch_Add tool as described in this forum 10 years ago: https://forum.freecad.org/viewtopic.php?f=9&t=4275

See the behavior:

https://github.com/user-attachments/assets/b036b181-63ec-45f5-8216-cecceaed56c3

Error message:
```
Running the Python command 'Arch_Add' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\BIM\bimcommands\BimArchUtils.py", line 56, in Activated
    FreeCADGui.doCommand("Arch.addSpaceBoundaries( FreeCAD.ActiveDocument."+sel[-1].Name+", FreeCADGui.Selection.getSelectionEx() )")
  File "<string>", line 1, in <module>

module 'Arch' has no attribute 'addSpaceBoundaries'
```

Should this be reported as a separate issue? As I stated previously, I am not sure if my steps are correct.

Roy-043, 2025-01-07:
There is indeed a regression which is caused by the ensureBase code. See #18864.

Additionally there is an issue if you select the top face of Slab001. I suspect this is a co-planar issue.

maxwxyz, 2025-01-07:
@paullee0 FYI, could break this if backported

paullee0, 2025-01-08:
I haven't looked at Arch Objects other than Wall and Stairs about the ensureBase() regression.

Indeed, as Roy_043 indicates, it causes similar regression in ArchSpace, could apply similar fix as done for Wall and Stairs but it seems from semhustej's video, there maybe some related issues.

Anyway, so for 'ensureBase() regression' :- 
- Wall :  Fixed
- Stairs : Fixed
- Space : confirmed, to be fixed
-  etc. ?

Hope there are other more tests, thanks.

paullee0, 2025-01-08:
PR to fix this submitted, anyone could test?  Thanks.

https://github.com/FreeCAD/FreeCAD/pull/18950

maxwxyz, 2025-01-08:
@paullee0 thanks, seems to fix it!

maxwxyz, 2025-01-08:
@paullee0 not sure if this has worked before but it fails/creates a wrong space where 2 walls meet at an angle like this:
[image]

Even like this, where I have not intersecting boundaries:
[image]

Maybe it has troubles if the angle is >270 deg?

paullee0, 2025-01-08:
I think it is the current algorithm limitation like 'concave' case Yorik mentioned.

That's why I made some improvement with a few videos below (oh, 3 years ago) how Spaces could be made automatically without such problems :

https://www.youtube.com/watch?v=0bnQZ473R7A
https://www.youtube.com/watch?v=Ob78XVOnmXQ
https://www.youtube.com/watch?v=1XR1I9WG75Y
https://www.youtube.com/watch?v=6E_rgq762Q8

Should have discussion in FC forum ...

maxwxyz, 2025-01-09:
sounds great!

closed 2025-01-10 by yorikvanhavre