FreeCAD/FreeCAD #18921: Images clipped during PDF export

mpfusion, 2025-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During PDF export parts of the drawing are clipped and the template title block is over-written. The SVG export works, though.

Steps to reproduce:
- create a simple drawing
- open TD and create default page
- click Insert Active View and position image close to title block
- add another Active View next to the existing one
- Export to PDF

PDF Export
[image]

SVG Export
[image]

[clipped_image.tar.gz](https://github.com/user-attachments/files/18333704/clipped_image.tar.gz)
[export.pdf](https://github.com/user-attachments/files/18333705/export.pdf)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39792 (Git)
Build type: Release
Branch: main
Hash: 278ce803907ef72548d7ac761613038ac435c481
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.35
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-07:
How do you export as PDF?

mpfusion, 2025-01-08:
@maxwxyz Right click on the page → Export PDF

maxwxyz, 2025-01-08:
@WandererFan FYI

WandererFan, 2025-01-08:
My first guess is that this is related to using the PDF/A-1b variant of the pdf spec.  It does not allow transparency and ActiveView sets the background of the captured image to transparent.   We had a similar problem with the transparent fill for balloons.

Not sure about the solution.  Maybe we need a property to control which pdf variant we use.

WandererFan, 2025-01-08:
see #15692 for background on svg clipping and Qt6.

mpfusion, 2025-01-21:
I don't know the PDF/SVG internals so I'm not sure if the following suggestion is even possible, but one way to solve this without using transparency would be to *add* the images. Or or logical OR them: A∨B

That way white parts in the image stay white. And the order in which the images are placed on the template wouldn't matter.

mpfusion, 2025-01-29:
My current workaround: Hit CTRL-P → Print to File (PDF) on each individual page and print those individual pages to pdf: 01.pdf, 02.pdf, etc. Then use `pdftk *.pdf cat output final.pdf`.

Not great and a lot of manual work, but if someone needs a workaround, here it is.

luzpaz, 2025-02-01:
Is this a dupe of #15692

WandererFan, 2025-02-02:
Not the same as #15692 - that is about svg-tiny not supporting clipping within an svg file.

This one is about a pdf version that does not allow transparency.  Left to itself PDF/A-1b will replace transparent areas with a black fill.  This looks terrible, so we replace transparent fill with page colour before export.  The page color area blocks whatever is behind it.   It is more obvious if you use a non-transparent background on the ActiveView.

tetektoza, 2025-04-12:
@WandererFan how about creating a Dialog window or anything like that, or triggering separate panel on "Tasks" window with PDF format selection? I just replaced our Pdf version from A-1b to X4 and it works correctly, here is the result:
[image]

How about this dialog/panel, or we can change it to X-4 by default? I can do it, but let me know on what we would want to decide here. BTW, if I remember correctly - X-4 is more intended to be used for our use cases.

WandererFan, 2025-04-12:
I think I'd go with a preference for pdf version.   That would work for headless use as well as gui.

Is pdf version something that a typical user would change often?  If that is a decision that needs to be made for every print, then the dialog might be better.

tobiasfalk, 2025-04-12:
I don't think that it is something that is a setting that needs changing with every print but having a option to change the version would be nice, since there may be cases where the user needs to have a specific version.

It is sadly in German that this describes the difference rather good:
https://www.fast-report.com/de/blogs/comparison-pdf-formats

Of what I understand, PDF/E would actually be the version that is theoretically ideal for FreeCAD since it is meant for engendering.

Also, these are the Versions supported by  QT:
- PDF 1.4
- PDF A-1B
- PDF 1.6
- pDF X4

tetektoza, 2025-04-12:
> I think I'd go with a preference for pdf version. That would work for headless use as well as gui.
> 
> Is pdf version something that a typical user would change often? If that is a decision that needs to be made for every print, then the dialog might be better.

Do you think it would also be suitable to put it the option to choose (could be a combobox) in Edit -> Preferences? I see that some people could use it, not sure if @tobiasfalk necessarily, but maybe someone could choose themselves.

tobiasfalk, 2025-04-12:
Yes having it in the options would definitely be a good solution, maybe in import/export or something. 
If one could also adjust other things for the PDF Page there it would be ideal but not necessari.

tobiasfalk, 2025-04-12:
I also think that the PDF export most likely needs much rework, especially if 3D models( #12843 )  should be included  in the future. This is poorly based on the fact that QT PDF print is in my experience a bit buggy in generall.

tetektoza, 2025-04-12:
@tobiasfalk I'd add those options for now, but in the future if the rework will come, we can start rethinking this feature incrementally. How about this? I can also support PDF for a little bit since I've had some previous experience.

tobiasfalk, 2025-04-12:
@tetektoza do they only affect the PDF export for TD or do they effect all pdf exports? I think there where two or three different ones.

tetektoza, 2025-04-12:
> [@tetektoza](https://github.com/tetektoza) do they only affect the PDF export for TD or do they effect all pdf exports? I think there where two or three different ones.

Not sure how to enable transparent layer (like Active View in TD in other tools), but I'd guess other printers or writers that are in FC have the same problems as well, as I can see we only use `A1b` for the paint device everywhere.

tobiasfalk, 2025-04-12:
Ok, thx

closed 2025-04-28 by chennes