FreeCAD/FreeCAD #18927: 10:26:29  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!

colt45s1911a1, 2025-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

i'm having issues where im able to constrain some lines but not others, then i tried constraining a fresh line of a rectangle after i deleted the previous line because i was not able to constrain the previous line and i got this error. 
10:26:29  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!
there are no other constraints on this line or the parallel line, so i have no idea how i have a redundant constraint.
can you post a link so i can go back to the previous version until this version is stable, please???

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.0
  * sheetmetal 0.3.13
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
Please add an example file and steps to reproduce

luzpaz, 2025-03-21:
@Tadzik1977 no file was attached