FreeCAD/FreeCAD #18932: Tree overlay interferes with 3D manipulation performance

dzid26, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If tree overlay is active, and the mouse was hovering over the tree recently, the panning and orbiting is being interrupted / choppy for a while. 
Sometimes I see the tree is still reacting to hovering while panning the 3D view. If I stop the mouse, or if I start circling over the tree a 1 CPU core is freed and jerkiness disappears.

https://github.com/user-attachments/assets/9c74a143-7a0b-4321-8ef9-f6ee43cbb936

For some reason during recording, the effect wouldn't last long.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * CubeMenu (Disabled)
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * kicadStepUpMod 11.4.4
  * Launcher
  * ModernUI (Disabled)
  * Ondsel-defaults.cfg
  * OpenTheme 2024.9.1
  * PieMenu 1.10.1
  * POV-Ray-Rendering
  * ToolbarStyle
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
@Rexbas FYI. Is it the same/similar to https://github.com/FreeCAD/FreeCAD/issues/17185?

Rexbas, 2025-01-08:
I cannot reproduce this

dzid26, 2025-01-08:
I was able to reproduce on a fresh 1.0.0  install.  This machine is a bit better and tearing was only happening on a 4K display and not on FullHD anymore. I could not notice CPU increase because it has more / faster cores.  
Perhaps if the machine is fast enough it becomes harder to reproduce. 

The initial machine was Intel 11th gen.
The second one was Ryzen Z1 Extreme.  Both have integrated GPUs.

obelisk79, 2025-01-16:
I am unable to reproduce.

dzid26, 2025-01-22:
> @Rexbas FYI. Is it the same/similar to #17185?

No, happens even with preselection and selection turned off and even for a very simple models. 

It has something to do with defocusing overlay panel widgets. 
I would expect some lost frames when the panel switch to transparent, but here, if I keep orbiting/panning, the tearing lasts few seconds or sometimes until I stop moving.  

(again no idea why but the tearing lasts <1s when recording!!! ):

https://github.com/user-attachments/assets/6e03650b-6e79-4f21-a7bc-936731938ef9 

BUT it behaves much better (not perfect, but <1s of tearing) if I replace 
https://github.com/FreeCAD/FreeCAD/blob/ee2f327a964d93239c0cf6fbb8ca813a76123e4b/src/Gui/OverlayWidgets.cpp#L1459
with `_graphicsEffect->setEnabled(false)`
Or in the therme `Dark Theme + Dark Background.qss`: `qproperty-enableEffect: 0;`

luzpaz, 2025-08-31:
Is this a duplicate of #12620

dzid26, 2025-09-03:
@luzpaz  I tested with [2025.09.03](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.09.03)  I don't think https://github.com/FreeCAD/FreeCAD/issues/12620 talks about performance at all.

In 1.1 the issue with the tree is somewhat more obvious.  

It is now very clear there is something wrong with tree text highlights - they are very laggy, the CPU core goes high and there are even some transparency artifacts.  Especially on 4K monitor. Look at this lag:

https://github.com/user-attachments/assets/d2667f72-b981-4022-ae19-c45e20843b02

It still affects the model rotation but only _sometimes_ it lasts for few seconds after leaving the tree area. 

Now, again, setting Gui--OverlayTabWidget -> `qproperty-enableEffect: 0` to zero solves the issue. 

https://github.com/user-attachments/assets/f15def69-3e72-4c30-add3-12b598ccb1e3

luzpaz, 2025-09-03:
@dzid26 can you upload the model for us to test with ?

dzid26, 2025-09-03:
I think any model with multiple tree items.  
Stock AssemblyExample will do:

https://github.com/user-attachments/assets/6596e8b1-7496-4c8b-b315-cdd879bdd7ee