FreeCAD/FreeCAD #18933: User-specified location macros are missing

ein-shved, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After applying #18525, users starts reporting that their macros from user specified location are missing

### Full version info

```shell
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53

OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ein-shved, 2025-01-08:
@maxwxyz H! Can you assign this to me, please

ein-shved, 2025-01-08:
@Syres916 @tarman3 @macdroid53, FYI

tarman3, 2025-01-08:
Maybe this is important

FreeCAD 1.1.0dev.39786 not show any macros in folder,
but I can see list of **Recent macros** and can run macros by hotkeys Ctrl+Shift+N  

[image]