FreeCAD/FreeCAD #18936: Section view hatches are rasterized during export

mpfusion, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a drawing with a section view hatch is exported the hatch is rasterized. This happens with PDF as well as SVG export. Both formats are vector formats and capable of rendering vector hatches.

Steps to reproduce:
- create a simple drawing
- open Techdraw
- insert default page and drawing
- insert simple section view to create a hatch
- Right click → Export PDF

If the hatch is created using “Apply geometric hatch to Face” the hatch is exported correctly.

GUI:
[image]

Export:
[image]

[hatch.pdf](https://github.com/user-attachments/files/18345686/hatch.pdf)
[hatch.tar.gz](https://github.com/user-attachments/files/18345688/hatch.tar.gz)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39792 (Git)
Build type: Release
Branch: main
Hash: 278ce803907ef72548d7ac761613038ac435c481
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.35
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2025-01-08:
This is a known issue with svg hatch on export.  We convert them to images because otherwise they are not clipped to the shape of the face\. Qt QSvgRenderer does not support clipping as it is not part of the svg-tiny spec.

There is some talk about this being fixed in the Qt6 development cycle.

WandererFan, 2025-01-08:
See the comments on #15692 for background on svg clipping in Qt.

mpfusion, 2025-01-08:
But is also an issue with PDF export.

mpfusion, 2025-01-09:
And one more observation that's quite unexpected: the export does not only rasterize the exported file but also the TechDraw view:

https://github.com/user-attachments/assets/59319b20-27a0-4797-96a0-c376e96b5dea

The current view should be unaffected by merely exporting.

WandererFan, 2025-01-09:
`The current view should be unaffected by merely exporting.`
A recompute will put the view back to vector graphics.   

`But is also an issue with PDF export.`
The hatches are svg items.  Even when exporting to pdf the Qt svg code that draws the hatches will not trim them to fit the face.

FEA-eng, 2025-09-20:
In my case, it just becomes solid after export and the view is not exported for some reason:

[image]

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

davidosterberg, 2026-05-10:
Still relevant