FreeCAD/FreeCAD #18937: Regression with Facebinder -> 2D Offset ("Wire is not closed") ?

byte-for-byte, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When doing: 
Shape from text -> Select ShapeString -> Part/Extrude -> Select top face of Extrude object -> Draft/Facebinder -> Select Facebinder object -> Part/2D Offset 
results in: 
"Wire is not closed"

This workflow works for really simple text, like the letter "I". The letter "S" does not work and throws the error
(font used: Ubuntu_B.ttf).

AFAIK this worked in 0.18. I cannot test it since the app images < 1.0 do not run on my linux box.

### Full version info

```shell
OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-08:
Please attach FCStd file and font

byte-for-byte, 2025-01-08:
Yep sorry! :)
[files.zip](https://github.com/user-attachments/files/18346168/files.zip)

tarman3, 2025-01-08:
Offset not working with bspline

byte-for-byte, 2025-01-08:
I dont want to introduce extra software tools in a workflow which should IMO work in the 3d CAD tool. Several other tools can do this easily.

Also this recommendation does not explain why this obviously worked (and I quote YT video and experience by other useres) in previous versions.

I was therefore not necessarily looking for a workaround, but rather trigger a discussion whether this is a regression or not. I thought this is the right spot for this plan.

BTW: I created svgs in inkscape and imported them. trying to extrude them gives very varying results. Some created broken extrusions. I don't think that this is the way to go...

Thank you for the input anyway!

tarman3, 2025-01-08:
Please write version of FreeCAD which do correct offset for Shapestring with bspline

maxwxyz, 2025-01-08:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/10435 / https://github.com/FreeCAD/FreeCAD/issues/18836

closed 2025-01-08 by maxwxyz

byte-for-byte, 2025-01-09:
hi tarman,
thanks for the reply!

please check out this video, the second part where he extrudes the text 'ing'.

https://youtu.be/yUVWQxITn1o?si=nf5pswp23NHb-seR

may i ask i you are a FC dev?

thanks

tarman3, 2025-01-09:
Agree, it is regression
Probably in video used FreeCAD 0.19

I checked version 0.21.2. Offset not working.
So this feature was broken for a long time

PS I am not developer, just an ordinary user

byte-for-byte, 2025-01-09:
Hi Tarman,
thanks you. Good to see that somebody else corroborates my regression theory.
I wonder why such regressions wouldn't be captured by (unit)tests?

I also wonder, why the ticket was closed by maxwxyz, giving the reason "not planned".
I do not understand why a previously working great feature broke at some point and it is not planned to restore the functionality?

Btw: I asked you if you are a dev, b.c. I was curious if FC-devs actually suggest to use another tool. ;)

Well, I am also working software dev, and my plan was to support making FC better by investing my time in thoroughly testing v1.x and reporting back my findings here.

I could already open the next bug ticket here about importing simple csv from inkscape that break extrusion in the Parts WS. But given the time it takes to test things, build a MWE and submit everything here, and not getting any useful feedback.... I wonder if it is worth it... maybe it is too early to work with FC...?

tarman3, 2025-01-09:
Given the current state of FreeCAD, if the problem can be circumvented, then it is not a problem

tarman3, 2025-01-09:
>resulted in another extrusion failure when extruding a simple line drawing

I want see, Please attach this file

JohnOCFII, 2025-01-09:
> I also wonder, why the ticket was closed by maxwxyz, giving the reason "not planned".

If you read the text at this link https://github.com/FreeCAD/FreeCAD/issues/18937#issuecomment-2577932252, this issue was closed because it is a duplicate of other noted issues. (I think there is some sort of automatic tag that shows every closed issue as not planned - but the actual reason/description was manually entered above the GitHub generated close message. )