FreeCAD/FreeCAD #18940: TechDraw: Spreadsheet view upscaled in 1.1.0dev

furgo16, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When compared to 1.0.0, in 1.1.0dev Spreadsheet views in TechDraw pages appear upscaled. A workaround is to manually set the view's `Scale` property to something like 0.35 (from the default 1.00). However, this makes drawings incompatible between the FreeCAD 1.0.0 and 1.1.0dev versions.

## Steps to reproduce

1. Download and uncompress 
[TestSheet.zip](https://github.com/user-attachments/files/18346869/TestSheet.zip), open it with FreeCAD 1.0.0. The file was created with FreeCAD 1.0.0
2. Check the size and scale of the `Sheet` view in the document tree. Scale is 1.0 and size appears to be correct => OK
1. Open the same file uncompressed in step 1, but open it with FreeCAD 1.1.0dev.
2. Check the size and scale of the `Sheet` view in the document tree. Scale is 1.0 and size appears to be oversized => NOK

FreeCAD 1.0.0:
[image]

FreeCAD 1.1.0dev
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39792 (Git) Snap 1254
Build type: Release
Branch: main
Hash: 278ce803907ef72548d7ac761613038ac435c481
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Assembly4 0.50.15
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2025-01-08:
Spreadsheet views are actually svg symbols under the covers.  We recently fixed the way svg is scaled and I guess this is an unwanted side effect.   There is a parameter that controls using the new accurate scaling vs the old, inaccurate but good for spreadsheets.

[image]

WandererFan, 2025-01-08:
FWIW, I'm leaning towards replacing the current spreadsheet view.   The TD code mimics a lot of Spreadsheet code to convert each cell to an svg representation.

Looks like it is possible to have Spreadsheet module serve up an HTML representation of the sheet that could be loaded directly into a text item in TD.

maxwxyz, 2025-01-08:
@WandererFan should this issue be closed then?

furgo16, 2025-01-09:
I've added a note about this behavior at https://wiki.freecad.org/Release_notes_1.1#TechDraw_Workbench

For future reference, this seems to be the PR that fixed the SVG symbol scaling: https://github.com/FreeCAD/FreeCAD/pull/18757

closed 2025-02-09 by WandererFan