FreeCAD/FreeCAD #18941: build: Import: missing include on build

matsievskiysv, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When building a subset of modules, Import module compilation fails to build with error

```
In file included from /tmp/build/src/Mod/Import/App/ExportOCAF.cpp:46:
/tmp/build/src/App/Document.h:38:10: fatal error: QString: No such file or directory
   38 | #include <QString>
      |          ^~~~~~~~~
compilation terminated.
```

The selected modules are: `IMPORT`, `PART`, `PART_DESIGN`, `SKETCHER`, `SPREADSHEET`.

Problem is not observed when building the default set of modules or `IMPORT` alone.

Build may be reproduced using the following file and command

<details>

<summary><code>freecad.dockerfile</code></summary>

```dockerfile
FROM debian:bookworm

ARG DEBIAN_FRONTEND=noninteractive
ARG BUILD_DIR=/tmp/build

RUN apt-get update -q \
    && apt-get install -qy --no-install-recommends \
    build-essential \
    cmake \
    git \
    cmake \
    cmake-extras \
    libboost-date-time-dev \
    libboost-dev \
    libboost-filesystem-dev \
    libboost-graph-dev \
    libboost-iostreams-dev \
    libboost-program-options-dev \
    libboost-python-dev \
    libboost-regex-dev \
    libboost-serialization-dev \
    libboost-thread-dev \
    libcoin-dev \
    libdouble-conversion-dev \
    libfmt-dev \
    libglew-dev \
    liblz4-dev \
    libopencv-dev \
    libeigen3-dev \
    libgtest-dev \
    libgmock-dev \
    libgl-dev \
    libgts-bin \
    libgts-dev \
    libkdtree++-dev \
    libmedc-dev \
    libmetis-dev \
    libmsgsl-dev \
    libocct-data-exchange-dev \
    libocct-ocaf-dev \
    libocct-visualization-dev \
    libpyside2-dev \
    libqt5opengl5-dev \
    libqt5svg5-dev \
    libqt5xmlpatterns5-dev \
    libqt5x11extras5-dev \
    libshiboken2-dev \
    libspnav-dev \
    libvtk9-dev \
    libx11-dev \
    libxerces-c-dev \
    libyaml-cpp-dev \
    libzipios++-dev \
    lsb-release \
    nlohmann-json3-dev \
    occt-draw \
    pybind11-dev \
    pyqt5-dev-tools \
    pyside2-tools \
    python3-dev \
    python3-matplotlib \
    python3-pivy \
    python3-ply \
    python3-pyside2.qtcore \
    python3-pyside2.qtgui \
    python3-pyside2.qtsvg \
    python3-pyside2.qtuitools \
    python3-pyside2.qtwidgets \
    python3-pyside2.qtxml \
    python3-requests \
    qtbase5-dev \
    qttools5-dev \
    qtwebengine5-dev \
    swig \
    xauth \
    xvfb \
    && apt-get autoremove \
    && apt-get autoclean \
    && rm -rf /var/lib/apt/lists/*

ARG FREECAD_URL=https://github.com/FreeCAD/FreeCAD.git
ARG FREECAD_COMMIT=c42031690377122a80ff574ef596b2a6cec5adbf
RUN mkdir ${BUILD_DIR} \
    && cd ${BUILD_DIR} \
    && git init --initial-branch=master \
    && git remote add origin ${FREECAD_URL} \
    && git fetch origin ${FREECAD_COMMIT} \
    && git reset --hard FETCH_HEAD \
    && git submodule update --init --recursive

RUN cd ${BUILD_DIR}\
    && cmake -S . -B build \
    -DCMAKE_BUILD_TYPE=Release \
    -DBUILD_TESTING=OFF \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DBUILD_GUI=ON \
    -DFREECAD_BUILD_DEBIAN=ON \
    -DBUILD_WITH_CONDA=OFF \
    -DINSTALL_TO_SITEPACKAGES=ON \
    -DBUILD_QT5=ON \
    -DBUILD_FEM=OFF \
    -DBUILD_SANDBOX=OFF \
    -DBUILD_TEMPLATE=OFF \
    -DBUILD_ADDONMGR=OFF \
    -DBUILD_BIM=OFF \
    -DBUILD_DRAFT=OFF \
    -DBUILD_DRAWING=OFF \
    -DBUILD_HELP=OFF \
    -DBUILD_IDF=OFF \
    -DBUILD_IMPORT=ON \
    -DBUILD_INSPECTION=OFF \
    -DBUILD_JTREADER=OFF \
    -DBUILD_MATERIAL=OFF \
    -DBUILD_MESH=OFF \
    -DBUILD_MESH_PART=OFF \
    -DBUILD_FLAT_MESH=OFF \
    -DBUILD_OPENSCAD=OFF \
    -DBUILD_PART=ON \
    -DBUILD_PART_DESIGN=ON \
    -DBUILD_CAM=OFF \
    -DBUILD_ASSEMBLY=OFF \
    -DBUILD_PLOT=OFF \
    -DBUILD_POINTS=OFF \
    -DBUILD_REVERSEENGINEERING=OFF \
    -DBUILD_ROBOT=OFF \
    -DBUILD_SHOW=OFF \
    -DBUILD_SKETCHER=ON \
    -DBUILD_SPREADSHEET=ON \
    -DBUILD_START=OFF \
    -DBUILD_TEST=OFF \
    -DBUILD_MEASURE=OFF \
    -DBUILD_TECHDRAW=OFF \
    -DBUILD_TUX=OFF \
    -DBUILD_WEB=OFF \
    -DBUILD_SURFACE=OFF \
    -DBUILD_VR=OFF \
    -DBUILD_CLOUD=OFF \
    -DENABLE_DEVELOPER_TESTS=OFF \
    && cmake --build build -j$(nproc) \
    && cmake --install build \
    && cd / \
    && rm -rf ${BUILD_DIR}
```

</details>

```bash
docker build -f freecad.dockerfile --network=host --tag freecad:latest $(mktemp -d)
```

### Full version info

```shell
[The latest main branch commit.](https://github.com/FreeCAD/FreeCAD/commit/c42031690377122a80ff574ef596b2a6cec5adbf)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2025-01-08 by matsievskiysv

MarcusMWilliams, 2025-01-08:
Meaning of closed? Please .

MarcusMWilliams, 2025-01-08:
Marcus Williams confirmation

matsievskiysv, 2025-01-08:
> Meaning of closed? Please .

When building, I did not take into account warning messages about module dependencies. After enabling dependencies for selected modules error disappeared.