FreeCAD/FreeCAD #18943: Duplication constraint when i use the dimension tool

Reapper, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i select a wire to set a constraint, Freecad set directly and duplicate this same constraint when i move the mouse.

### Full version info

```shell
A video is better than a long text. 
https://github.com/user-attachments/assets/03ceeed7-8560-4540-8d6f-d82ae27a6481

I tested this issue on the released version and the dev version. That never happened before the version 1.0.0 release

- Version 1.0.0 About:
OS: Ubuntu 22.04.5 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.50
  * A2plus 0.4.65
  * fasteners 0.5.24
  * Help 1.0.3
  * Assembly4 0.50.13
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.18
  * CurvedShapes 1.0.9

- Version 1.1.0dev About:
OS: Ubuntu 22.04.5 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39793 (Git)
Build type: Release
Branch: main
Hash: c42031690377122a80ff574ef596b2a6cec5adbf
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.50
  * A2plus 0.4.65
  * fasteners 0.5.24
  * Help 1.0.3
  * Assembly4 0.50.13
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.18
  * CurvedShapes 1.0.9

Linux version :
Linux user 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
@PaddleStroke @FlachyJoe @AjinkyaDahale have you ever seen such a behavior?

Reapper, 2025-02-18:
Hello, it's seem solved. 
I did a test on version 1.0.0 recently and i can't redo the issue, the constraint tool work fine and doesn't duplicate them.
So... it's probably a wrong version python module or something like that, cause i did an `apt update` few days ago.