FreeCAD/FreeCAD #18944: unchecking "ESC can leave Sketch edit mode" does not work

MartinMajewski, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I really dislike the behavior with which FreeCad leaves a sketch after pressing the escape key one time to often. Sometimes it would make life easier to hit the escape key multiple times to deselect a tool and/or all selections inside a sketch.

I found the option "ESC can leave Sketch edit mode" in the applications preferences under the Sketcher menu. However, deselecting this option does not seem to have an effect. FreeCad leaves the sketch edit mode with the escape key anyways.

[image]

### Full version info

```shell
OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2025-01-08:
I am not able to reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

MartinMajewski, 2025-01-08:
However, it shows a systemic issue with how some changes are applied. It seems to me that each Sketcher object holds that attribute of how it can be dismissed but is not actively updated by the settings system. It needs to read these settings attributes from a global store on instantiation during the first creation or project opening.

This could be "important at that moment" if other settings that might influence a workflow are not immediately applied, and you notice it only on restart when everything goes boom. It should at least be noticed and evaluated by the dev team.

tarman3, 2025-01-08:
Probably message "Please restart to apply changes" will be appropriate

MartinMajewski, 2025-01-08:
It might help to make this behavior less confusing.

maxwxyz, 2025-01-08:
Seems like it is saved per sketch.
@PaddleStroke @AjinkyaDahale FYI

AjinkyaDahale, 2025-01-29:
The problem seems to be in `../Gui/TaskDlgEditSketch.cpp:53`:
```
setEscapeButtonEnabled(hGrp->GetBool("LeaveSketchWithEscape", true));
```
This may be being called only when starting the dialog is created, and not updated while it is on.

The relevant commit is c01c1f7fde6851219d4b5ed38d906024be58e890 by @wwmayer. Pinging if he could shed some light.

fallingcats, 2025-04-11:
I found the Esc key doesn't work in FreeCAD at all anymore, it does not close context menus, or work in the menu bar either.

luzpaz, 2025-05-09:
soft bump

PR #22049 (2025-06-30): Sketcher: fix Esc can leave setting synchronization
The "Esc can leave sketch edit mode" setting was working, but it's state wasn't properly refreshed after changes (requiring a restart for it to take effect).

All the required code was already there just a small mistake in property name. Seems like the wrong name was there from the time it was introduced in 92e6094 . I am guessing it was a placeholder/copy paste error which wasn't properly tested due to being part of giant refactoring commit.

Test plan:
* enable "esc can leave sketch edit mode" sketcher option
* restart FreeCad (just to be s [... cut]