FreeCAD/FreeCAD #18947: BIM_Shape2DCut does not produce cut, only view

semhustej, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- open file 
[shape-2d-view.FCStd.zip](https://github.com/user-attachments/files/18350433/shape-2d-view.FCStd.zip)

- select the Section object
- start BIM_Shape2DCut
- the resulting object does not have the correct setting - the `Projection mode` is set to `Solid` instead of `Cutfaces`

[image]

plus there is an error message:
```
Running the Python command 'BIM_Shape2DCut' failed:
Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39786-conda-Windows-x86_64-py311\Mod\Draft\draftguitools\gui_shape2dview.py", line 72, in Activated
    self.proceed()
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39786-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimShape2DView.py", line 102, in proceed
    FreeCADGui.doCommand("sv.ProjectionMode = \"Cutfaces\"")
  File "<string>", line 1, in <module>

Cannot access attribute 'ProjectionMode' of deleted object
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

The issue has been discovered by another user in this forum thread: https://forum.freecad.org/viewtopic.php?t=93675

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
I get the same output.
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * parts_library
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5

```

yorikvanhavre, 2025-01-09:
Looks like a bug :) I'll look into it

ErtugrulEge, 2025-01-19:
In the latest weekly, still projection mode property is solid but error has gone also, if you select "cut faces" option, you can use Draft-Pattern property but pattern size option is  limited.I tried for larger patterns but  I couldn't do it on a larger scale.

Last thing, in the feature we will need an alignment degree option. For example you need to align the brick pattern accoording to wall.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

[image]

paullee0, 2025-01-29:
Seems 'sv' is not passed from Shape2dView to Shape2dSection, should be a simple fix ...

`        FreeCADGui.doCommand("sv.ProjectionMode = \"Cutfaces\"")`

paullee0, 2025-01-29:
PR Submitted - https://github.com/FreeCAD/FreeCAD/pull/19316

paullee0, 2025-01-29:
> ...  if you select "cut faces" option, you can use Draft-Pattern property but pattern size option is limited.I tried for larger patterns but I couldn't do it on a larger scale.
>
> Last thing, in the [...]

Maybe open another one or two issue on these separately :)

closed 2025-02-28 by yorikvanhavre