FreeCAD/FreeCAD #18952: BIM: Arch_Space label cannot be rotated

maxwxyz, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the label is displayed on an Arch_Space, it is always rotated so it is viewable on the XY-Plane.
It is not possible to rotate the label, neither around Z on the XY-Plane not orient it to a different plane.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * parts_library
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
Could be a separate issue but I guess should be solved with this one:
When Location is (0,0,0) the label is positioned at the bottom face.
When you enter a z coordinate (e.g. 10) to increase the height where the display is displayed, it is displayed at (0,0,10).
IMO it should be still automatic for X and Y - otherwise it would need a height input possibility to change the location.

yorikvanhavre, 2025-01-09:
> IMO it should be still automatic for X and Y

That's a good point, indeed, and should be easy to do.

I think the text position should be changed from a vector to a placement, so it can have full rotation, etc. There is code to handle that in BuildingPart.py, maybe we can reuse it here. I'll have a look

maxwxyz, 2025-03-04:
Besides the proposed solution, this could also use a similar mechanism like the attachment offset, basically providing a matrix to offset the label by position and rotation from the default position.