FreeCAD/FreeCAD #18955: FreeCAD v0.21 is unmaintained

Zolko-123, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD v0.21 was compatible with every version from 0.19, meaning that there are many installed instances of FreeCAD v0.19 to 0.21, which share the same compatible file and data format. It is possible to open a model made with 0.19 / 0.20 / 0.21 in any other 0.19 / 0.20 / 0.21 version. 

During the v0.22 development period, many changes changes to the data format – including TNP fixes – make models made with pre-0.21 and post-0.22 incompatible, meaning it's not possible to reliable open and share documents between those versions. There are also discussion about new file formats for v1.1 in the future ... For existing long-term projects, it is therefore very important that the FreeCAD v0.21 series continues to be available in the future. For existing professional projects it is fundamental to be able to trust the continuous availability of that former version, and that it continues to be supported for the foreseeable future.

Unfortunately, that stable series is unmaintained since years, and is not accessible through the website's download page either, although it's going to be part of current distributions (Debian 12 for example) for years.

I ask therefore that a new maintainer for the v0.21 be named, to ensure continuous development of that last compatible version. I'm sure that there are a number of adequate candidates for that task, and I also propose my candidacy to be the maintainer of the 0.21 version.

FOSDEM at the end of january could be a good moment to discuss this problem

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.3.33852 (Git)
Build type: Compiled
Branch: v0.21
Hash: c8c307eb6bc5951ed05b324e6af26f29e6c7fced
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * A2plus 0.4.56a
  * CurvedShapes 1.0.3
  * fcgear 1.0.0
  * sheetmetal 0.2.50
  * TabBar
  * Render 2022.2.0
  * Curves 0.5.1
  * fasteners 0.5.21
  * Assembly4_dev 0.50.16-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-08:
I guess anyone can fork it...

hyarion, 2025-01-12:
@Zolko-123 from the developer meeting on 2024-11-16:

3. **Maintenance of 0.21 after 1.0?** (yorik)
   - Several people (zolko, syres IIRC) asked to still support 0.21 for a time after 1.0 is released.
   - Concerns about lack of manpower to do that maintenance
   - No objections in principle: if someone submits a PR with a backport, it will get reviewed

Zolko-123, 2025-01-13:
> if someone submits a PR with a backport, it will get reviewed

"_reviewed_ " by whom ? There is no maintainer

yorikvanhavre, 2025-01-13:
> "reviewed " by whom ? There is no maintainer

Any code contribution is reviewed and merged by the [same people](https://github.com/orgs/FreeCAD/teams/maintainers) (usually during monday meetings), no matter the release or branch...

We all collectively "maintain" FreeCAD. There is no designated "maintainer" for any release. The code - and builds - of each previous release since 0.13 is available under https://github.com/FreeCAD/FreeCAD/releases It is true that older releases are not directly advertised on the main download page at https://www.freecad.org/downloads.php but they are easy to find anyway, I don't have the impression we need more there...

Now, about maintaining 0.21, that doesn't necessarily means something specific. Will the 0.21 code stay available? Yes, of course. Will it still compile forever on modern OSes? Yes, but in the long term of course not, and it might be out of our reach to do anything. For example, modern distros will drop Qt5 at some point, and there would be no sense in porting 0.21 to Qt6. That's precisely why there is a 1.0 release... When will that happen? I have no idea, I imagine that depends on how many projects that use Qt5 (AppImages might work for much longer, though).

So it is not clear what is expected from the FreeCAD devs when you say "maintain 0.21". I think what we can agree that if some problem arises, and someone needs 0.21 to still work in a particular setup, of course we the maintainers will do our best to help. But there will be a moment when that becomes just way too much work to be reasonable.

hyarion, 2025-01-13:
> > if someone submits a PR with a backport, it will get reviewed
> 
> "_reviewed_ " by whom ? There is no maintainer

My understanding is that it would be same people and process as all other PRs.

For example if syres creates a PR, you can review it (or me or anyone else). Then yorik, chennes, wwmayer or any other maintainer of the freecad project/repository can merge it.

maxwxyz, 2025-01-13:
Can this issue be closed then?

Zolko-123, 2025-01-13:
> So it is not clear what is expected from the FreeCAD devs when you say "maintain 0.21"

Who are "the FreeCAD devs" ? Since I have contributed code to FreeCAD, I'm obviously one of them, but that doesn't make sense in the context of your phrase. Could you please explain who or what "FreeCAD devs" are ?

According to me, this is the very summary of what's wrong with the current Ondsel-imposed development process : you make a distinction between "developers" and "users". You divide the world in a us –vs– them. Whereas I can observe that most "developers" are incompetent when it comes to CAD because they don't seem to be "users". While many "users" tend to contribute some code to FreeCAD thus becoming "developers". 

So if I go to the root of your question, then the only good answer is : "nothing", I want FreeCAD **USERS** to be in charge of FreeCAD v0.21 and I expect the _FreeCAD devs_ to focus on v1.0 and then 1.1 and 1.2 and ...

yorikvanhavre, 2025-01-13:
Honestly don't know what more to reply here...

furgo16, 2025-01-13:
> Who are "the FreeCAD devs"?

I think what it was meant on that sentence was "maintainers". The current list of maintainers and their role is documented publicly at https://freecad.github.io/DevelopersHandbook/maintainersguide.

This is not explaining anything new, that is how most open source projects work. A team of maintainers have write access to the repository and are in charge of merging contributions from those who submit code. Everyone can submit their PRs, add functionality and fix issues.

> According to me, this is the very summary of what's wrong with the current Ondsel-imposed development process : you make a distinction between "developers" and "users". You divide the world in a us –vs– them. 

As a FreeCAD user who has contributed code, and who is not a maintainer, I don't see an imposed division. This is a common way of running an open source project.

maxwxyz, 2025-01-13:
I'm closing this. If there is further need for discussions I can convert it to a discussion.

Zolko-123, 2025-01-31:
> I guess anyone can fork it...

strictly speaking is FreeCAD v1.0 the fork as it's not compatible with the 20 previous years of FreeCAD, and has also changed the contributors team

For those who wish to continue to use the long-term data format of v0.21, I'll keep it maintained for the time being : 

https://github.com/Zolko-123/FreeCAD_0.21