FreeCAD/FreeCAD #18956: Segfault when adding offset to shapestring subshapebinder

kefir-, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I wanted to offset a ShapeString, but it resulted in a segfault. To reproduce:

1. Part Design: Create Body and Sketch, draw a simple rectangle and create a Pad
2. Draft: add ShapeString, place it on the Pad
3. Part Design: Create a SubShapeBinder of the ShapeString, which brings Binder object inside Body
4. Try to add 1mm offset in Binder -> Data -> Offset. This reports a segfault in the console of FreeCAD
5. Set offset back to 0mm, and then try to set it to 1mm again. Now FreeCAD crashes completely.

Attaching segfault.FCSTD, which is ready for step 4. Just set offset to 1, then 0, and then 1 again.

Discovered in FreeCAD v1.0.0, reproduced in latest weekly build.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kefir-, 2025-01-08:
Same issue reported on freecad forum, with sample FCStd attached:

https://forum.freecad.org/viewtopic.php?t=93703

xtemp09, 2025-01-10:
Perhaps, I'll leave the backtrace here.

<details><summary>Backtrace</summary>

```cpp
(gdb) bt
#0  Bisector_BisecAna::SetTrim (this=0x0, uf=0.33510360229338748, ul=0.59319354740747499)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/Standard/Standard_Handle.hxx:130
#1  0x00007fffc9edd372 in MAT2d_Tool2d::BisecFusion (this=<optimized out>, I1=<optimized out>, 
    I2=<optimized out>) at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/MAT2d/MAT2d_Tool2d.cxx:1200
#2  0x00007fffc9eed933 in BRepMAT2d_BisectingLocus::RenumerationAndFusion (this=this@entry=0x7fffffffa580, 
    ILine=ILine@entry=3, LengthLine=30, IndexLast=@0x7fffffff9df8: 72, NewMap=...)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepMAT2d/BRepMAT2d_BisectingLocus.cxx:198
#3  0x00007fffc9ef17a5 in BRepMAT2d_BisectingLocus::Compute (this=this@entry=0x7fffffffa580, anExplo=..., 
    IndexLine=IndexLine@entry=1, aSide=aSide@entry=MAT_Left, aJoinType=GeomAbs_Arc, IsOpenResult=false)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepMAT2d/BRepMAT2d_BisectingLocus.cxx:149
#4  0x00007fffc86382ea in BRepFill_OffsetWire::Init (this=this@entry=0x7fffffffa4f0, Spine=..., 
    Join=Join@entry=GeomAbs_Arc, IsOpenResult=IsOpenResult@entry=false)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepFill/BRepFill_OffsetWire.cxx:399
#5  0x00007fffc86385cd in BRepFill_OffsetWire::BRepFill_OffsetWire (this=<optimized out>, Spine=..., 
    Join=<optimized out>, IsOpenResult=<optimized out>, this=<optimized out>, Spine=..., 
    Join=<optimized out>, IsOpenResult=<optimized out>)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepFill/BRepFill_OffsetWire.cxx:358
#6  0x00007fffd27ced8d in BuildDomains(TopoDS_Face&, NCollection_List<TopoDS_Shape>&, NCollection_List<BRepFill_OffsetWire>&, GeomAbs_JoinType, bool, bool, bool&) [clone .constprop.0] (myFace=..., WorkWires=..., 
    myAlgos=..., myJoin=GeomAbs_Arc, myIsOpenResult=false, isPositive=isPositive@entry=true, 
    isWasReversed=@0x7fffffffacfe: true)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepOffsetAPI/BRepOffsetAPI_MakeOffset.cxx:277
#7  0x00007fffd2713479 in BRepOffsetAPI_MakeOffset::Perform (this=0x7fffffffb290, Offset=1, Alt=0)
    at /usr/src/opencascade-7.6.3+dfsg1-7.1build1/src/BRepOffsetAPI/BRepOffsetAPI_MakeOffset.cxx:330
#8  0x00007fffd071a85b in Part::BRepOffsetAPI_MakeOffsetFix::Perform (this=this@entry=0x7fffffffb240, 
    Offset=Offset@entry=1, Alt=Alt@entry=0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Part/App/BRepOffsetAPI_MakeOffsetFix.cpp:99
#9  0x00007fffd07c0e64 in Part::TopoShape::makeElementOffset2D (this=this@entry=0x7fffffffb980, shape=..., 
    offset=offset@entry=1, joinType=joinType@entry=Part::JoinType::arc, 
    fill=fill@entry=Part::FillType::noFill, 
    allowOpenResult=allowOpenResult@entry=Part::OpenResult::noOpenResult, 
    intersection=intersection@entry=false, op=0x7fffd08219ab "OFF")
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Part/App/TopoShapeExpansion.cpp:2763
#10 0x00007fffb1b4ab67 in Part::TopoShape::makeElementOffset2D (this=this@entry=0x7fffffffbc80, offset=1, 
    join=join@entry=Part::JoinType::arc, fill=fill@entry=Part::FillType::noFill, 
    allowOpenResult=allowOpenResult@entry=Part::OpenResult::noOpenResult, 
    intersection=intersection@entry=false, op=0x0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Part/App/TopoShape.h:1024
#11 0x00007fffb1b3aa33 in PartDesign::SubShapeBinder::update (this=this@entry=0x555559fb7450, 
    options=options@entry=PartDesign::SubShapeBinder::UpdateForced)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/PartDesign/App/ShapeBinder.cpp:811
#12 0x00007fffb1b3b81d in PartDesign::SubShapeBinder::execute (this=0x555559fb7450)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/PartDesign/App/ShapeBinder.cpp:877
#13 0x00007ffff5136bf5 in App::DocumentObject::recompute (this=0x555559fb7450)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/DocumentObject.cpp:167
#14 0x00007fffd0553fda in Part::Feature::recompute (this=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Part/App/PartFeature.cpp:108
#15 0x00007ffff50af7f2 in App::Document::_recomputeFeature (this=this@entry=0x5555587ed610, 
    Feat=0x555559fb7450) at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Document.cpp:3469
#16 0x00007ffff50b7c7e in App::Document::recompute (this=this@entry=0x5555587ed610, 
    objs=std::vector of length 0, capacity 0, force=force@entry=false, hasError=hasError@entry=0x0, 
    options=options@entry=0) at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Document.cpp:3177
#17 0x00007ffff623a878 in Gui::PropertyEditor::PropertyEditor::recomputeDocument (
    this=this@entry=0x555555d4b0b0, doc=0x5555587ed610)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/propertyeditor/PropertyEditor.cpp:374
#18 0x00007ffff623aa44 in Gui::PropertyEditor::PropertyEditor::closeTransaction (
    this=this@entry=0x555555d4b0b0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/propertyeditor/PropertyEditor.cpp:400
#19 0x00007ffff623b496 in Gui::PropertyEditor::PropertyEditor::closeEditor (this=0x555555d4b0b0, 
    editor=<optimized out>, hint=QAbstractItemDelegate::SubmitModelCache)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/propertyeditor/PropertyEditor.cpp:423
#20 0x00007ffff2b12dbf in doActivate<false> (sender=0x555555f9a3a0, signal_index=4, argv=0x7fffffffc8d0)
[... cut ...]
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2206
#41 0x00007ffff5eae5d5 in (anonymous namespace)::runEventLoop (mainApp=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2235
#42 0x00007ffff5eb248d in Gui::Application::runApplication ()
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2315
#43 0x000055555556a293 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Main/MainGui.cpp:295
(gdb) 
```

</details>

luzpaz, 2025-01-10:
Is this a PD or Draft issue ?

xtemp09, 2025-01-10:
@luzpaz, OCCT.

FEA-eng, 2025-11-01:
I can't reproduce it on Windows.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251031 (Git shallow)
Build date: 2025/10/31 03:23:50
Build type: Release
Branch: main
Hash: c270a8ac666dff22a92baf2a3a4f5200197bb678
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

pieterhijma, 2026-10-27:
I cannot reproduce on FreeCAD `main` with OCCT 8.1 dev:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

Changing offsets works as expected, so I assume the issue has been resolved in OCCT. Closing.

FEA-eng, 2026-10-27:
> Changing offsets works as expected, so I assume the issue has been resolved in OCCT.

I don't have a crash, but I get:

`19:57:39  Binder: failed to make 2D offset`

when setting offset to 1.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

luzpaz, 2026-10-28:
> I cannot reproduce on FreeCAD main with OCCT 8.1 dev  
> ...
> Changing offsets works as expected, so I assume the issue has been resolved in OCCT. Closing.

@pieterhijma, the OCCT Liason issue metadata was missing for what version of OCC this was fixed in. I added it. 

Should this issue be closed since @FEA-eng responded above with:

> I don't have a crash, but I get:
> `19:57:39  Binder: failed to make 2D offset`
> when setting offset to 1.

Edit: I'm re-opening issue so it doesn't get lost. I also suggest to alter the title of the ticket from
`OCC: Segfault when adding offset to shapestring subshapebinder`  
to  
`OCC: (Was: Segfault) Warning when adding offset to shapestring subshapebinder`

closed 2026-11-06 by pieterhijma