FreeCAD/FreeCAD #18962: Adding Leading Zeros to Numerical Inputs Causes Incorrect Number to be Inputted

TheTechnicGuy1, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Typing in whole numbers with leading zeros (not decimals) the whatever-er-er is not truncating the zeros off the front. Weirdly it doesn't seem to matter how many leading zeros there are (tested up to 5 leading zeros with two different numbers) and it always spits out the same, albeit wrong result. This bug seems to be for 'all' numerical inputs (uh yeah ok I didn't test all of them but it seems to exist for most of the operations, sketch shapes, and variable sheets, as well as it crashes the settings panel), and the problem persists in safe mode. 

42 = 42
but...
042 = 34
0042 = 34
00042 = 34
ect. ect.
069 = 6
0420 = 272

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.35
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-09:
Zero at the beginning indicates octal system
042 (8) = 34 (10)

Resolved in dev build

Syres916, 2025-01-09:
@Roy-043 would it be possible to please backport https://github.com/FreeCAD/FreeCAD/commit/19fa91253fbcaae7d8786a0fc85d83e31d615bb7 to the 1.0 branch as it was missed at the time probably because the author didn't specifically ask.

Roy-043, 2025-01-09:
The mentioned commit has been backported to V1.0.1.

closed 2025-01-09 by Roy-043

Syres916, 2025-01-09:
@Roy-043 thanks.

@TheTechnicGuy1 I've just tested the backport on what will become version 1.0.1 and it works so just be patient until it's released and no I don't have an ETA, sorry.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39158 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: e557658def7b6721ce1c97ea3d44b3daeb63cb9e
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.3
  * sheetmetal 0.6.13
  * Behave-Dark-Colors 0.1.1
  * Alternate_OpenSCAD 1.0.0
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.34
  * Curves 0.6.53
  * Assembly4 0.50.15
  * A2plus 0.4.68

```