FreeCAD/FreeCAD #18963: Holes create spurious handles

mpfusion, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Holes create spurious handles which make it hard to hit the correct one:

https://github.com/user-attachments/assets/b40e28f5-e4c7-4d99-9508-80f0315eefed

Steps to reproduce:
- create a simple part
- add hole
- import part into TechDraw with view from the top/bottom

Result:
[image]

Expected result:
[image]

These spurious handles often conceal the important ones, which makes it hard to correctly dimensions angles:
[image]

[spurious handles.tar.gz](https://github.com/user-attachments/files/18359962/spurious.handles.tar.gz)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39792 (Git)
Build type: Release
Branch: main
Hash: 278ce803907ef72548d7ac761613038ac435c481
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.35
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mpfusion, 2025-01-09:
I forgot to add the TechDraw label and cannot find a way to add it after the fact.

FEA-eng, 2025-01-09:
The vertex dots are shown for convenience when creating dimensions and can be disabled along with the frames: https://wiki.freecad.org/TechDraw_ToggleFrame

There's a preference to set their size too.

> I forgot to add the TechDraw label and cannot find a way to add it after the fact.

Choose TechDraw as "Subproject(s) affected?":

```
### Subproject(s) affected?

TechDraw
```

maxwxyz, 2025-01-11:
@mpfusion What would be the expected outcome? Just the start and end points and the conical line in between (I guess it would be a spline)?
@WandererFan is this possible?

mpfusion, 2025-01-11:
@maxwxyz The expected outcome would be to have just three handles like in the 2nd picture “Expected result”. One handle for each end and one for the centre, instead of the fiveteen or so I'm currently getting.