FreeCAD/FreeCAD #18964: Text Justification change during export

Industrial-Designer, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello ,

why my rotated texts path justification change from center to left after export it as dxf ? because after import it i found texts shifted

how to keep it FIX ?

thanks in advance

### Full version info

```shell
it back end
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-09:
Try to use legacy exporter
[image]

maxwxyz, 2025-01-11:
@Industrial-Designer please add full about info and an example file.

closed 2025-01-20 by yorikvanhavre

Industrial-Designer, 2025-02-26:
@maxwxyz I try to draw a protractor so text need to be align with marks , I draw it as wanted but after export always getting text justification change to left and mine is center so all texts are shifted

PR #19096 (2025-01-20): Draft: DXF export: add handling of text justification
Fixes #18964.
commit ddd86bd7 (2025-01-20): Draft: DXF export: add handling of text justification

Fixes #18964.