FreeCAD/FreeCAD #18966: Github doesn't allow upload of native FreeCAD files

mpfusion, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Github rejects uploading of FreeCAD files with the error message:
> We don’t support that file type.
>
> Try again with GIF, JPEG, JPG, MOV, MP4, PNG, SVG, WEBM, CPUPROFILE, CSV, DMP, DOCX, FODG, FODP, FODS, FODT, GZ, JSON, JSONC, LOG, MD, ODF, ODG, ODP, ODS, ODT, PATCH, PDF, PPTX, TGZ, TXT, XLS, XLSX or ZIP.

[image]

It's an unfortunate choice to reject native files for the program this very bug tracker is for. A workaround is to create a `.tar.gz` file (note that an uncompressed `.tar` is not enough, it has be be a compressed archive to be accepted) and include the FreeCAD file in the archive.

It would make sense to configure github to allow `.FCStd` files. Users are supposed to upload a minimal example file with the bug reports. And currently github unfortunately rejects those files.

### Full version info

```shell
none
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-09:
> It would make sense to configure github to allow .FCStd files. Users are supposed to upload a minimal example file with the bug reports. And currently github unfortunately rejects those files.

AFAIR, this is not possible (ATM). The change would need to happen on the Github side, right?

mpfusion, 2025-01-09:
I don't know. Maybe it's configurable in the admin panel (which I hope it is). In case it's not: raise an issue with Github. It makes no sense to reject native FreeCAD files on a FreeCAD bug tracker.

tarman3, 2025-01-09:
Just packing FCStd to zip

mpfusion, 2025-01-09:
@tarman3 A quick search suggests that Windows supports `.tar.gz` natively. But that's off-topic here. I just wanted to offer a workaround. There are probably many more workarounds to trick Github into accepting unsupported files. But none of those should be necessary. That's what this report is about.

maxwxyz, 2025-01-09:
> Just add ".NOT.zip" to the end of the file name before uploading it. GH will treat it as a zipped file (which it is).

It is always a surprise when unzipping an example file. Great if the .FCStd was zipped but not so ideal if it was renamed 😄

yorikvanhavre, 2025-01-10:
> @chennes @yorikvanhavre is it possible to allow different (custom) file types in the repository?

Not that I know of... AFAIK this is all entirely managed by GitHub. We see people asking for other extensions all the time (ex https://github.com/orgs/community/discussions/42494 ) but doesn't look like we would have a lot of chances to be heard :sweat_smile:

closed 2025-01-10 by maxwxyz