FreeCAD/FreeCAD #18968: Crashes in sketch edit

PaulPestell, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am running FreeCAD in Xubuntu, I am getting random crashes usually when using polyline. The programme just vanishes. I not very familiar with using the terminal commands but tried running from terminal by typing in FreeCAD but it came back 'command not found'.  
I tried it in safe mode and it happened as I was dimensioning a constraint to construction geometry. Shall I just uninstall and reinstall FreeCAD. 
 I am considering trying Blender, but I am an engineer not an Artist Ha Ha!!!  
Thanks in anticipation.  PS I am not a developer, just a 74 year old ex engineer trying to learn a great CAD programme.  

### Full version info

```shell
OS: Ubuntu Core 22 (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-09:
Try with last weekly build
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

PaulPestell, 2025-01-09:
Thanks for the info, but it diddnt work.  Note, I downloaded the build then double clicked it and it opened FreeCAD.  Hope this was right.

PaulPestell, 2025-01-09:
Sorry closed by mistake.
I managed to open FreeCAD in terminal.  I got a massive page of red code on terminal after it crashed.  
This is the first few lines

Adding snap-specific PYTHONPATH to sys.path: /home/paul/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1248/lib/python3.10/site-packages:/snap/freecad/1248/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f6e0d689520]
#1  /snap/freecad/1248/kf5/usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so(+0x3c7b64) [0x7f6e049a0b64]
#2  /snap/freecad/1248/kf5/usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so(+0x2f8072) [0x7f6e048d1072]

maxwxyz, 2025-01-11:
Does the crash happen with all files or a specific one? Are there steps how you can reliably reproduce it?

PaulPestell, 2025-01-11:
Hi Max, thanks for your reply, there was nothing consistent. I have
re-installed freecad and hoped that it would overwrite the error. Go far
it’s working fine. I couldn’t uninstall because terminal diddnt recognize
‘freecad’, now I have two freecad launch icons 😂😂. I am about to install
a better desktop sidebar so I can get rid of the faulty command.
I am going to try and get better at using the terminal.
Thanks again for your help.

closed 2025-01-11 by maxwxyz