FreeCAD/FreeCAD #18969: Sketches loose their attachment when loading an older file

NomAnor, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a file that was created with a development version between 0.21 and 1.0. I don't know what commit I used.
When I open that file in 1.0, all Sketches (and only Sketches, not DatumPlanes for example) have no attachment supports.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +469 (Git)
Build type: Release
Branch: integration
Hash: 585d91c0e923b028b4d7476c6a21a861373d7969
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * ThreadProfile 1.84.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.3.0
  * fasteners 0.5.27
```

### Subproject(s) affected?

Sketcher

### Anything else?

I suspect that the "Attachment Support" property gets cleared by some update function after loading. I tried poking around in the AttachExtension code, but haven't found anything.

This is the xml extract for one of the sketches:
```
<Object name="Sketch" Extensions="True">
            <Extensions Count="1">
                <Extension type="Part::AttachExtension" name="AttachExtension">
                </Extension>
            </Extensions>
            <Properties Count="18" TransientCount="0">
                <Property name="AttacherType" type="App::PropertyString" status="8">
                    <String value="Attacher::AttachEnginePlane"/>
                </Property>
                <Property name="AttachmentOffset" type="App::PropertyPlacement">
                    <PropertyPlacement Px="0.0000000000000000" Py="0.0000000000000000" Pz="0.0000000000000000" Q0="0.0000000000000000" Q1="0.0000000000000000" Q2="0.0000000000000000" Q3="1.0000000000000000" A="0.0000000000000000" Ox="0.0000000000000000" Oy="0.0000000000000000" Oz="1.0000000000000000"/>
                </Property>
                <Property name="AttachmentSupport" type="App::PropertyLinkSubList">
                    <LinkSubList count="1">
                        <Link obj="XY_Plane001" sub=""/>
                    </LinkSubList>
                </Property>
                <Property name="Constraints" type="Sketcher::PropertyConstraintList">
                    <ConstraintList count="11">
                        <Constrain Name="" Type="1" Value="0.0000000000000000" First="0" FirstPos="2" Second="1" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="1" Value="0.0000000000000000" First="1" FirstPos="2" Second="2" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="1" Value="0.0000000000000000" First="2" FirstPos="2" Second="3" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="1" Value="0.0000000000000000" First="3" FirstPos="2" Second="0" SecondPos="1" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="2" Value="0.0000000000000000" First="0" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="2" Value="0.0000000000000000" First="2" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="3" Value="0.0000000000000000" First="1" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="3" Value="0.0000000000000000" First="3" FirstPos="0" Second="-2000" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="14" Value="0.0000000000000000" First="3" FirstPos="1" Second="3" SecondPos="2" Third="-1" ThirdPos="1" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="13" Value="0.0000000000000000" First="0" FirstPos="2" Second="-4" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                        <Constrain Name="" Type="13" Value="0.0000000000000000" First="0" FirstPos="2" Second="-3" SecondPos="0" Third="-2000" ThirdPos="0" LabelDistance="10.0000000000000000" LabelPosition="0.0000000000000000" IsDriving="1" IsInVirtualSpace="0" IsActive="1" />
                    </ConstraintList>
[... 57 lines cut ...]
                <Property name="Placement" type="App::PropertyPlacement" status="8388612">
                    <PropertyPlacement Px="0.0000000000000000" Py="0.0000000000000000" Pz="0.0000000000000000" Q0="0.0000000000000000" Q1="0.0000000000000000" Q2="0.0000000000000000" Q3="1.0000000000000000" A="0.0000000000000000" Ox="0.0000000000000000" Oy="0.0000000000000000" Oz="1.0000000000000000"/>
                </Property>
                <Property name="Shape" type="Part::PropertyPartShape">
                    <Part file="PartShape10.brp"/>
                </Property>
                <Property name="ShapeMaterial" type="Materials::PropertyMaterial">
                    <PropertyMaterial uuid="7f9fd73b-50c9-41d8-b7b2-575a030c1eeb"/>
                </Property>
                <Property name="Support" type="App::PropertyLinkSubList" status="67108864">
                    <LinkSubList count="0">
                    </LinkSubList>
                </Property>
                <Property name="Visibility" type="App::PropertyBool" status="648">
                    <Bool value="false"/>
                </Property>
            </Properties>
        </Object>
```

Comparing it to a newly saved file, the `AttacherEngine` property is missing. But this seems to be correctly fixed by the `restoreAttacherEngine` function. The `Attacher Engine` property is set to "Engine Plane".

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-09:
Could you share an example file?

NomAnor, 2025-01-09:
[SketchAttach.fcstd.zip](https://github.com/user-attachments/files/18365437/SketchAttach.fcstd.zip)

This files shows the behaviour, I don't have access to the version I used to create my original file.
The Sketch inside the Body should be attached to the XY-Plane as set by
```
<Property name="AttachmentSupport" type="App::PropertyLinkSubList">
    <LinkSubList count="1">
        <Link obj="XY_Plane" sub=""/>
    </LinkSubList>
</Property>
```
but because the file also contains
```
<Property name="Support" type="App::PropertyLinkSubList" status="67108864">
    <LinkSubList count="0">
    </LinkSubList>
</Property>
```
the code I linked above will overwrite it. Maybe the code should only overwrite the attachment support if the old `Support` list is non-empty.

FEA-eng, 2025-09-27:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

closed 2026-09-01 by chennes