FreeCAD/FreeCAD #18970: Draft Link Arrays throw error because of missing property

NomAnor, 2025-01-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. New File -> Parametric Part
2. Select Body
3. Create Draft Array with "Link Array" selected

The following error is thrown and not resulting shape is available:
```
Traceback (most recent call last):
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/array.py", line 64, in attach
    super(Array, self).attach(obj)
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/draftlink.py", line 83, in attach
    self.linkSetup(obj)
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/array.py", line 343, in linkSetup
    super(Array, self).linkSetup(obj)
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/draftlink.py", line 126, in linkSetup
    obj.setPropertyStatus('PlacementList', '-Immutable')
<class 'AttributeError'>: Property container has no property 'PlacementList'
Traceback (most recent call last):
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/array.py", line 439, in execute
    self.buildShape(obj, pl, pls)
  File "/home/xxx/src/FreeCAD/integration/build/debug/Mod/Draft/draftobjects/draftlink.py", line 191, in buildShape
    obj.setPropertyStatus('PlacementList', '-Immutable')
<class 'AttributeError'>: Property container has no property 'PlacementList'
Array: Property container has no property 'PlacementList'
```

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +469 (Git)
Build type: Release
Branch: integration
Hash: 585d91c0e923b028b4d7476c6a21a861373d7969
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * ThreadProfile 1.84.0
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.3.0
  * fasteners 0.5.27
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2025-01-09:
I can reproduce the issue.

I also tried with a body that is not empty and I get no error warning.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

Roy-043, 2025-01-09:
There is an error if you select an empty Body. But the error message I get is different.

```
18:02:42  pyException: Traceback (most recent call last):
  File "D:\BKG_Tmp\FreeCAD_1.1\Mod\Draft\draftobjects\array.py", line 439, in execute
    self.buildShape(obj, pl, pls)
  File "D:\BKG_Tmp\FreeCAD_1.1\Mod\Draft\draftobjects\draftlink.py", line 217, in buildShape
    raise RuntimeError(_err_msg)
<class 'RuntimeError'>: 'Array' cannot build shape from 'Body'

18:02:42  Array: 'Array' cannot build shape from 'Body'
```

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

NomAnor, 2025-01-09:
I get the error regardless if the Body is empty or not.

Roy-043, 2025-01-09:
I do not recognize your FreeCAD version.

Can you retest with an AppImage?

NomAnor, 2025-01-09:
I use a self compiled v1.0.0, I had to cherry pick a fix to compile it on my system, thats why the commit might be different.

Here is the error message with the official 1.0.0 AppImage:
```
19:13:34  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAgpOcFg/usr/Mod/Draft/draftobjects/array.py", line 64, in attach
    super(Array, self).attach(obj)
  File "/tmp/.mount_FreeCAgpOcFg/usr/Mod/Draft/draftobjects/draftlink.py", line 82, in attach
    obj.addExtension('App::LinkExtensionPython')
<class 'RuntimeError'>: Cannot add static property
19:13:34  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAgpOcFg/usr/Mod/Draft/draftobjects/array.py", line 439, in execute
    self.buildShape(obj, pl, pls)
  File "/tmp/.mount_FreeCAgpOcFg/usr/Mod/Draft/draftobjects/draftlink.py", line 191, in buildShape
    obj.setPropertyStatus('PlacementList', '-Immutable')
<class 'AttributeError'>: Property container has no property 'PlacementList'
19:13:34  Array: Property container has no property 'PlacementList'
```
I only see the `RuntimeError` in a release build.

semhustej, 2025-01-09:
I did not check the error message in my previous test. I get the same error as @Roy-043 with both 1.0 and 1.1 weekly.

I do not get the error when the body is not empty with both 1.0 and 1.1 weekly.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

NomAnor, 2025-01-09:
I can't explain it, but for some reason my user.cfg file seems to be messed up, causing this error.

If I start FreeCAD with an empty user.cfg (`-u user.cfg`), I can't reproduce the error. But if I run it with the user.cfg in my `.config` folder I get the previous error.
The user.cfg is bigger (50kB) than the newly created one (20kB).

Not sure if this is worth investigating any further, I will just use a newly generated file.

closed 2025-01-09 by NomAnor