FreeCAD/FreeCAD #18975: improved checks in GUI/Jobs.py/claimChildren

st68, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Error messages occur when a Job is not fully loaded (or, in my case, when not fully constructed via scripting).

### Full version info

```shell
I will provide tested changes.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

st68, 2025-01-10:
----------------------------------------------------------------
Stan Tomlinson (1):
      improved checks in GUI/Jobs.py/claimChildren

 src/Mod/CAM/Path/Main/Gui/Job.py | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/src/Mod/CAM/Path/Main/Gui/Job.py b/src/Mod/CAM/Path/Main/Gui/Job.py
index f9507401b6..7ad969d099 100644
--- a/src/Mod/CAM/Path/Main/Gui/Job.py
+++ b/src/Mod/CAM/Path/Main/Gui/Job.py
@@ -265,13 +265,14 @@ class ViewProvider:
 
     def claimChildren(self):
         children = []
-        children.append(self.obj.Operations)
+        if hasattr(self.obj, "Operations"):
+            children.append(self.obj.Operations)
         if hasattr(self.obj, "Model"):
             # unfortunately this function is called before the object has been fully loaded
             # which means we could be dealing with an old job which doesn't have the new Model
             # yet.
             children.append(self.obj.Model)
-        if self.obj.Stock:
+        if hasattr(self.obj, "Stock"):
             children.append(self.obj.Stock)
         if hasattr(self.obj, "SetupSheet"):
             # when loading a job that didn't have a setup sheet they might not've been created yet

st68, 2025-01-10:
[PullRequest.txt](https://github.com/user-attachments/files/18369075/PullRequest.txt)

st68, 2025-01-10:
improved checks in GUI/Jobs.py/claimChildren
    
      eliminate some error messages when a Job is not fully loaded
      (or when not properly constructed via scripting)
    
    TESTED ON V0.21 -- code hasn't changed since then

st68, 2025-01-10:
P.S. -- I know I'm doing this wrong -- apologies.  First time attempt...

st68, 2025-01-10:
> There is no Path workbench anymore. It's been renamed to CAM in v1.0.0
>
> Can you make a PR against main branch which is currently 1.1.0-dev ?

pulled from github to local server, reset to main, cherry-picked the
commit, and did request-pull -p.
Note there is no difference in the diff, other than the git indexes.
(I've not done a proper 'PR' before; if this isn't close enough, I'd be
happy to try again, with a few pointers.)

Stan Tomlinson (1):
      improved checks in GUI/Jobs.py/claimChildren

 src/Mod/CAM/Path/Main/Gui/Job.py | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/src/Mod/CAM/Path/Main/Gui/Job.py
b/src/Mod/CAM/Path/Main/Gui/Job.py
index 4499d992b6..e5f9e2f060 100644
--- a/src/Mod/CAM/Path/Main/Gui/Job.py
+++ b/src/Mod/CAM/Path/Main/Gui/Job.py
@@ -265,13 +265,14 @@ class ViewProvider:

     def claimChildren(self):
         children = []
-        children.append(self.obj.Operations)
+        if hasattr(self.obj, "Operations"):
+            children.append(self.obj.Operations)
         if hasattr(self.obj, "Model"):
             # unfortunately this function is called before the object has
been fully loaded
             # which means we could be dealing with an old job which
doesn't have the new Model
             # yet.
             children.append(self.obj.Model)
-        if self.obj.Stock:
+        if hasattr(self.obj, "Stock"):
             children.append(self.obj.Stock)
         if hasattr(self.obj, "SetupSheet"):
             # when loading a job that didn't have a setup sheet they might
not've been created yet

luzpaz, 2025-01-10:
@st68 thanks!

Couple of options: I can create the PR for you but there would be no attribution to you in the commit log (except in the summary) or you can follow the documentation[^1] on how to make PR.

[^1]: https://wiki.freecad.org/Source_code_management

luzpaz, 2025-01-10:
Cool, didn't notice that you had already created a PR before I replied. I've linked it to this ticket. Very much appreciate the contribution. We need some CAM folks to test this here.

closed 2025-01-27 by yorikvanhavre