FreeCAD/FreeCAD #18978: Selecting edges is really slow

Miksu80, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting edges, freeCAD stops for a long time before becoming active. It might have something to do with the measure tool and propably slow because of modeled threads.

[image]

File (prototyping so the features are not implemented the best possible way):
[flycutter.FCStd.zip](https://github.com/user-attachments/files/18371406/flycutter.FCStd.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Finnish/Finland (fi_FI)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * FEMbyGEN 2.3.0
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * Quetzal 1.0.1
  * sheetmetal 0.5.10
  * Silk 0.1.5
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2025-01-10:
I can confirm this issue.
It happens in my dev build:

```
OS: macOS Sonoma (14.5)
Architecture: arm64
Version: 1.1.0dev.39755 +1 (Git) Conda
Build type: Release
Branch: tmp2
Hash: 08d9be581e6242a9100310da82cd2c7b3fd9647e
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.6.11
```

It does not happen in 1.0

Pausing in execution in a debugger when this occurs gives me the following stacktrace:

```
...
    frame #21: 0x000000017100629c libTKTopAlgo.7.8.1.dylib`BRepBuilderAPI_GTransform::BRepBuilderAPI_GTransform(TopoDS_Shape const&, gp_GTrsf const&, bool) + 204
    frame #22: 0x0000000171fbc104 Part.so`Part::TopoShape::transformGShape(this=0x000000016fdfa6b8, rclTrf=0x000000016fdfac68, copy=false) const at TopoShape.cpp:3007:31
    frame #23: 0x0000000171fc5074 Part.so`Part::TopoShape::makeGTransform(this=0x000000016fdfa3a0, shape=0x000000016fdfa6b8, rclTrf=0x000000016fdfac68, op=0x0000000000000000, copy=false) at TopoShape.cpp:4163:20
    frame #24: 0x0000000171fbbdc8 Part.so`Part::TopoShape::makeGTransform(this=0x000000016fdfa6b8, mat=0x000000016fdfac68, op=0x0000000000000000, copy=false) const at TopoShape.h:815:28
    frame #25: 0x0000000171fbbc18 Part.so`Part::TopoShape::transformGeometry(this=0x000000016fdfa6b8, rclMat=0x000000016fdfac68) at TopoShape.cpp:2979:17
    frame #26: 0x0000000170468128 Measure.so`Measure::ShapeFinder::transformShape(inShape=0x000000016fdfabf8, placement=0x000000016fdfac10, scaler=0x000000016fdfac68) at ShapeFinder.cpp:197:12
    frame #27: 0x0000000170467ae4 Measure.so`Measure::ShapeFinder::getLocatedShape(rootObject=0x000000033f9c1000, leafSub="Pad004.;#282;:Hcd2,E.Edge59") at ShapeFinder.cpp:125:13
    frame #28: 0x000000017043fd20 Measure.so`Measure::Measurement::getShape(this=0x0000000168aacfb0, obj=0x000000033f9c1000, subName="Pad004.;#282;:Hcd2,E.Edge59") const at Measurement.cpp:284:12
...
```

<details><summary>Click for full stacktrace</summary>
<p>

```
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x0000000171737e4c libTKGeomBase.7.8.1.dylib`Extrema_GenExtPS::BuildGrid(gp_Pnt const&) + 880
    frame #1: 0x0000000171735cb4 libTKGeomBase.7.8.1.dylib`Extrema_GenExtPS::Perform(gp_Pnt const&) + 796
    frame #2: 0x000000017172852c libTKGeomBase.7.8.1.dylib`Extrema_ExtPS::Perform(gp_Pnt const&) + 636
    frame #3: 0x0000000171727ab4 libTKGeomBase.7.8.1.dylib`Extrema_ExtPS::Extrema_ExtPS(gp_Pnt const&, Adaptor3d_Surface const&, double, double, Extrema_ExtFlag, Extrema_ExtAlgo) + 296
    frame #4: 0x000000017168419c libTKGeomBase.7.8.1.dylib`___lldb_unnamed_symbol5155 + 1840
    frame #5: 0x00000001716e74e8 libTKGeomBase.7.8.1.dylib`AppCont_LeastSquare::FixSingleBorderPoint(AppCont_Function const&, double, double, double, NCollection_Array1<gp_Pnt2d>&, NCollection_Array1<gp_Pnt>&) + 500
    frame #6: 0x00000001716e9054 libTKGeomBase.7.8.1.dylib`AppCont_LeastSquare::AppCont_LeastSquare(AppCont_Function const&, double, double, AppParCurves_Constraint, AppParCurves_Constraint, int, int) + 5268
    frame #7: 0x00000001716be6c0 libTKGeomBase.7.8.1.dylib`Approx_FitAndDivide2d::Compute(AppCont_Function const&, double, double, double&, double&) + 552
    frame #8: 0x00000001716be128 libTKGeomBase.7.8.1.dylib`Approx_FitAndDivide2d::Perform(AppCont_Function const&) + 1452
    frame #9: 0x000000017168145c libTKGeomBase.7.8.1.dylib`ProjLib_ComputeApproxOnPolarSurface::ProjectUsingInitialCurve2d(opencascade::handle<Adaptor3d_Curve> const&, opencascade::handle<Adaptor3d_Surface> const&, opencascade::handle<Adaptor2d_Curve2d> const&) + 5592
    frame #10: 0x000000017167a47c libTKGeomBase.7.8.1.dylib`ProjLib_ComputeApproxOnPolarSurface::Perform(opencascade::handle<Adaptor2d_Curve2d> const&, opencascade::handle<Adaptor3d_Curve> const&, opencascade::handle<Adaptor3d_Surface> const&) + 460
    frame #11: 0x000000017167c73c libTKGeomBase.7.8.1.dylib`ProjLib_ComputeApproxOnPolarSurface::ProjLib_ComputeApproxOnPolarSurface(opencascade::handle<Adaptor2d_Curve2d> const&, opencascade::handle<Adaptor3d_Curve> const&, opencascade::handle<Adaptor3d_Surface> const&, double) + 104
    frame #12: 0x00000001707d1484 libTKBRep.7.8.1.dylib`BRepTools_NurbsConvertModification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) + 6048
    frame #13: 0x00000001707c9064 libTKBRep.7.8.1.dylib`BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) + 4096
    frame #14: 0x00000001707c89e8 libTKBRep.7.8.1.dylib`BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) + 2436
    frame #15: 0x00000001707c89e8 libTKBRep.7.8.1.dylib`BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) + 2436
    frame #16: 0x00000001707c89e8 libTKBRep.7.8.1.dylib`BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) + 2436
    frame #17: 0x00000001707c4e5c libTKBRep.7.8.1.dylib`BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) + 800
[... cut ...]
    frame #60: 0x0000000105986ef8 libFreeCADGui.dylib`(anonymous namespace)::runEventLoop(mainApp=0x000000016fdfe050) at Application.cpp:2235:9
    frame #61: 0x00000001059867bc libFreeCADGui.dylib`Gui::Application::runApplication() at Application.cpp:2315:5
    frame #62: 0x00000001000075dc FreeCAD`main(argc=3, argv=0x000000016fdfefd8) at MainGui.cpp:295:13
    frame #63: 0x00000001944ea0e0 dyld`start + 2360

```

</p>
</details> 

Which indicates that it is related to the new ShapeFinder code introduced in https://github.com/FreeCAD/FreeCAD/pull/18641

@WandererFan could you take a look at this?

Syres916, 2025-01-10:
Related to #18785

chennes, 2025-01-14:
Should be closed by https://github.com/FreeCAD/FreeCAD/pull/19032