FreeCAD/FreeCAD #18979: TechDraw: Insert View opens Select a SVG or Image  window

JoachimDunkel, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On Freecad 1.0 AppImage in Ubuntu 22.04

Situation:
1. Create any solid part
2.  Create a new drawing sheet with TechDraw workbench

Bug:
Selecting the body in tree and then pressing Insert View does not add the body but rather opens the Select a SVG view for me.
This happens wheiter I already have a page border existing or not.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

TechDraw

### Anything else?

I followed the tutorial in:

https://www.youtube.com/watch?v=VZNb4SpaKRk

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-10:
This isn't a bug, IMHO this is an educational scenario, 'drawing' is a different project document, the new page should be created in the same document as the Body and then a new View created in that page.

JoachimDunkel, 2025-01-10:
Thanks for clarifying the mistake.

From a Ui perspective it would have been much easier for me to find out if an error message like no Body in the current document ( selected ) would have poped up instead.

Then I would suggest changing the Issue to:
Insert View button does more then it should and behaves non-intuitive.

Do you agree ?

maxwxyz, 2025-01-10:
The above file dialog should only appear if you have nothing selected. Can you recheck that you have a solid selected before pressing the button?

Syres916, 2025-01-10:
For what it's worth, the `Insert View` icon should be disabled if the page of another document is in focus while the 2D/3D object in the original document is selected IMHO.

maxwxyz, 2025-01-10:
@Syres916 is this possible to differentiate? You should be able to click it without a selection but also if you have only the page or a sheet open. In that case I think the WB thinks nothing is selected (in that document)

Syres916, 2025-01-10:
> is this possible to differentiate?

Probably not right now but maybe the simple addition of an `objectName` under the hood so we can verify a `Page` belonging to which `document` doesn't sound beyond the realms of possibility.

luzpaz, 2025-11-23:
@JoachimDunkel can you test on weekly development build please ? https://github.com/FreeCAD/FreeCAD/releases

FEA-eng, 2026-09-03:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```