FreeCAD/FreeCAD #18980: Import: exported step model is scaled after gmsh meshing

matsievskiysv, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Exported step model is scaled by 1000 after gmsh meshing, when both are used in the same python session.

The following code imports some FreeCAD model, exports it to STEP format, meshes it in GMSH and then repeats the process (part of workflow for automated geometry optimization).

The file is exported scaled by 1000 after GMSH mesh generation.

Setting `Part.exportUnits` to different value does not help the situation.

```python
import FreeCAD
import Import
import Part
import gmsh
import gc

def show_line(file, start):
    with open(file, 'r') as f:
        for line in f.readlines():
            if line.startswith(start):
                print(line)
                break

doc = FreeCAD.openDocument('/usr/local/share/examples/PartDesignExample.FCStd')
body = doc.findObjects(Type='PartDesign::Body')[0]

Part.exportUnits('MM')
Import.export([body], '/tmp/out.step')
FreeCAD.closeDocument(doc.Name)
del body
del doc
gc.collect()

# inspect some point coordinates
show_line('/tmp/out.step', '#25 =')

gmsh.initialize()
gmsh.option.setString('Geometry.OCCTargetUnit', 'M')
gmsh.model.add('model')
gmsh.merge('/tmp/out.step')
gmsh.model.mesh.generate(2)
gmsh.write('/tmp/out.msh')
gmsh.finalize()

doc = FreeCAD.openDocument('/usr/local/share/examples/PartDesignExample.FCStd')
body = doc.findObjects(Type='PartDesign::Body')[0]

Part.exportUnits('MM')
Import.export([body], '/tmp/out.step')
FreeCAD.closeDocument(doc.Name)

# inspect the same point coordinates
show_line('/tmp/out.step', '#25 =')
```

### Full version info

```shell
Discovered in [Debian packaged v0.20.2](https://packages.debian.org/bookworm/freecad), reproduced in build from source v1.0.0 and [latest main branch commit](https://github.com/FreeCAD/FreeCAD/commit/558eb19103518090d494fcde0e8deabba830f734)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2025-01-14:
I'm experiencing 1000x scaling with OCCT 7.8.1, but not with OCCT 7.7.2.  I suspect that may be the underlying cause.

colt45s1911a1, 2025-01-30:
i'm having a similar problem when exporting. 

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.37
  * sheetmetal 0.7.5

[scale problem.zip](https://github.com/user-attachments/files/18607902/scale.problem.zip)