FreeCAD/FreeCAD #18981: Spreadsheet on a TechDraw page different size in 1.1dev

semhustej, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sample file:
[FC-11-sheet-scale.FCStd.zip](https://github.com/user-attachments/files/18376275/FC-11-sheet-scale.FCStd.zip)

When I open sample file in FreeCAD 1.1, the size of the sheet on the page is different than in 1.0 version  (where the file was created). The properties of the sheet seem to be the same.

Spreadsheet in FC 1.0:
[image]

Spreadsheet in FC 1.1dev
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2025-01-10:
@semhustej, #18940

semhustej, 2025-01-10:
Thanks for the information, closing this one as a duplicate of https://github.com/FreeCAD/FreeCAD/issues/18940

closed 2025-01-10 by luzpaz