FreeCAD/FreeCAD #18982: [Modification Request] Change Import Workflow

Mannshoch, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

According to: https://forum.freecad.org/viewtopic.php?t=92211

I sometimes import data into FreeCAD, mostly I do not need all data and mostly I struggle with the Import mode and have to do it more than once. This Request try to create a better UX, workflow and reduce waiting time.

Current Workflow:

1. Import
2. Chose file
3. Optional - change import settings
4. Import

Proposed Workflow:

1. Chose File
2. Analysing/basic Import happens in background. User could do other things in FreeCAD.
3. After finished a message appear and allow to enter the import mode.
4. Choose bodies/ surface / annotation (all preselected)
   A new view open with all bodies selected, a tree-view and a Task that allow to change Import-settings. According to the changed import settings in the Task the resulting view update.
   e.g. settings that change the tree, update the tree-view like the import mode for STEP files does.
5. _(Addons could hook in here and allow to add additional possibilities. Like repairing meshes or remove holes)_
6. Tick OK 
7. If Import into an already existing FreeCAD File happen. Show selected Bodies for placing. 
Other wise a new file is generated.
8. While user could place the Body, depending of changed settings a final processing or reimport happens in background and the import-objects are locked for changes (but not for placing).
9. After the background process finished processing the data the data got updated/replaced and the lock get removed

Assumed Problems:

- If you change settings that change how a body is imported a reimport may needed because an instant update is not possible. (
- converting tools may not designed for interaction. And could not help predict a file-structure if you change settings.

### Full version info

```shell
--
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-11:
@FreeCAD/cad-advisory-group @FreeCAD/design-working-group 
What is your opinion on the behavior?

pierreporte, 2025-01-12:
Basically there are two things here:

1. The import should be done in the background.
2. There should be some sort of pre-import to allow more options.

The first point is very interesting and it could be something used for pretty much any file opening: it only locks the relevant tab, or creates a new one when done, while allowing the user to continue working in other tabs. I feel like that #11914 can be a blocker for this.

The second point is also interesting but is it even possible? Can you extract the structure of a STEP file without loading everything else? Can you load only a subset of the file?

Mannshoch, 2026-09-14:
see also: https://github.com/FreeCAD/FreeCAD/pull/32466

This seems a good first release of this idea. It is limited to IFC and does not support a tree preview and not import only selected parts.

furgo16, 2026-09-14:
> The second point is also interesting but is it even possible? Can you extract the structure of a STEP file without loading everything else? Can you load only a subset of the file?

That's how we do it with the DXF import, IIRC. The file is even prescanned to count the number of entities, and a warning of a long import time is shown to the user, if it's a big file.