FreeCAD/FreeCAD #18985: Auto-constrained geometry needs two undo actions to be cancelled

pierreporte, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating sketch geometry with automatic constrains (horizontal or vertical line, circle with center on a line, etc.), it needs undoing twice to be removed: the first time it removes the constrains, the second time the geometry. Expected result: the geometry and its constrains are removed at once, with a single action.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * sheetmetal 0.6.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-10:
It could also be helpful in scenarios where you automatically placed a circle too close and want to get rid of the constraint only, without redrawing but I get it that it is not intuitive UX.

pierreporte, 2025-01-10:
Even then, all the constrains are removed by the first undo while only a subset of them could be wrong. It is already possible to select a constrain to delete it. The UX consensus on this is that things created implicitly should be removed with the main object. Following this principle will make sketcher more intuitive.

pierreporte, 2025-01-19:
Superseded by #19139

closed 2025-01-19 by pierreporte