FreeCAD/FreeCAD #18987: Pad creates null shape when going "up to face" in some geometry

Jillybread, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A pad operation fails when the diameter of the "up to face" geometry is below 130mm.
To replicate:

1. Open the attached file. Select "sketch002" and select the pad tool.
2. Pad the sketch "up to face" and select the outer diameter of the "BB" shape. The pad will succeed. Cancel the pad operation.
3. Open "sketch001" and change the outer diameter from 130mm to 128mm.
4. Repeat step 2. The pad should fail with a "null shape" error.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[bike.zip](https://github.com/user-attachments/files/18381679/bike.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-11:
@FlachyJoe FYI

FEA-eng, 2026-10-17:
Pad in the original file fails to recompute. A new one asks to enable "Allow Compound".

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```