FreeCAD/FreeCAD #18991: techdraw incorrectly shows thor shape

kazan417, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

techdraw incorrectly shows thor shape
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Plot 2024.11.26
```

### Subproject(s) affected?

TechDraw

### Anything else?

chosen Russian language

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kazan417, 2025-01-11:
[bug.zip](https://github.com/user-attachments/files/18384367/bug.zip)
freecad project file inside zip

maxwxyz, 2025-01-11:
can you test if this issue is also present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds