FreeCAD/FreeCAD #18993: Sketcher Trim tool only works on first selection, subsequent selections produce error

macdroid53, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have searched and there are multiple sketcher trim issues.
When using the Trim tool in Sketcher, the first edge is trimmed as expected. The tool is still active, but will no longer trim.
Each subsequent click on an edge produces the following:
(On Debian)

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.CADKernelError'>: Geom_TrimmedCurve::U1 == U2
App.getDocument('Unnamed').getObject('Sketch').trim(6,App.Vector(20.112181,9.455127,0))
Failed to trim edge
`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39806 (Git)
Build type: Release
Branch: main
Hash: 7829fcc4888bce067fe002a4d857a4953d2482fc
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * CurvedShapes 1.0.13
  * Silk 0.1.5
  * OpenTheme 2024.9.1
  * Curves 0.6.53
  * lattice2 1.0.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

macdroid53, 2025-01-11:
On Manjaro it produces the following error:
`08:13:12  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.CADKernelError'>: {'sclassname': 'N4Base14CADKernelErrorE', 'sErrMsg': 'Geom_TrimmedCurve::U1 == U2', 'sfile': '/var/tmp/pamac-build-mac/freecad-git/src/FreeCAD/src/Mod/Part/App/Geometry.cpp', 'iline': 2374, 'sfunction': 'virtual void Part::GeomTrimmedCurve::setRange(double, double)', 'swhat': 'Geom_TrimmedCurve::U1 == U2', 'btranslatable': False, 'breported': False}
08:13:12  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').trim(5,App.Vector(-6.000002,9.520015,0))
08:13:12  Unnamed#Sketch: Failed to trim edge
`

`OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39806 (Git)
Build type: Release
Branch: main
Hash: 7829fcc4888bce067fe002a4d857a4953d2482fc
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * ThreadProfile 1.92.0
  * DynamicData 2.70.0
  * OpenTheme 2024.9.1
  * workfeature-macro
  * Defeaturing 1.2.3
  * Curves 0.6.53
  * Silk 0.1.5
  * CurvedShapes 1.0.13
  * fasteners 0.5.35
  * kicadStepUpMod 11.4.4
  * Manipulator 1.5.7
  * lattice2 1.0.0
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.6.13
`

macdroid53, 2025-01-11:
Steps to reproduce:

- create a sketch
- create overlapping geometry
- select Trim tool
- click a segment of an edge, trim completes
- click another segment, no trim happens, shows errors as noted

depthoffocus, 2025-01-11:
I think this is probably #18497.

macdroid53, 2025-01-11:
Probably, I didn't catch that one when I searched.

maxwxyz, 2025-01-11:
Duplicate of #18497