FreeCAD/FreeCAD #18994: Incorrect gcode.  Gcode isn't completing final depth instruction

Lin1Jay1, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have searched 'incorrect gcode.  I got 1 hit https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20incorrect%20gcode&type=code    but I don't think it relates.

The problem, put simply, is the attached FreeCAD file is calling for 2 dowel holes to a depth of -0.390 in. (-9.906mm) in 0.130 in. (3.302 mm) steps using a 6mm end mill.  The gcode generates the the Z= -3.302 step, then the Z= -6.604 step but never generates the Z=-9.906 step required.  
If I increase the total depth by 0.001" (or 0.0001) it generates the Z=-9.906 step but it doesn't generate the last step to get down the last 0.001 in.  The work around seems to be to make steps equal to (total depth / an integer = to steps desired) +0.0001"
I had an identical problem yesterday in a different file and I just worked around it.  But it looks like this could be an issue: gcode isn't completing final depth instructions.

[Gcode Depth problem, 0.39inat0.13incr..FCStd.zip](https://github.com/user-attachments/files/18386134/Gcode.Depth.problem.0.39inat0.13incr.FCStd.zip)

(Z Problem -2 Dowel Holes x 0.332 x 0.390,  0.000 pass ext)
G0 Z7.620
G0 X-227.745 Y142.937
G0 Z3.810
G1 X-227.745 Y142.937 Z-3.302 F762.000
G2 X-227.662 Y141.318 Z-3.302 I-0.855 J-0.855 K0.000 F1143.000
G2 X-229.363 Y141.143 Z-3.302 I-0.938 J0.763 K0.000 F1143.000
G2 X-229.538 Y142.845 Z-3.302 I0.763 J0.938 K0.000 F1143.000
G2 X-227.745 Y142.937 Z-3.302 I0.938 J-0.763 K0.000 F1143.000
G1 X-227.745 Y142.937 Z-6.604 F762.000
G2 X-227.662 Y141.318 Z-6.604 I-0.855 J-0.855 K0.000 F1143.000
G2 X-229.363 Y141.143 Z-6.604 I-0.938 J0.763 K0.000 F1143.000
G2 X-229.538 Y142.845 Z-6.604 I0.763 J0.938 K0.000 F1143.000
G2 X-227.745 Y142.937 Z-6.604 I0.938 J-0.763 K0.000 F1143.000
G0 Z7.620
G0 Z7.620
G0 X-227.745 Y142.937
G0 X227.617 Y-141.467
G0 X227.617 Y-141.467 Z3.810
G1 X227.617 Y-141.467 Z-3.302 F762.000
G2 X227.701 Y-141.350 Z-3.302 I0.983 J-0.614 K0.000 F1143.000
G2 X229.499 Y-142.813 Z-3.302 I0.899 J-0.731 K0.000 F1143.000
G2 X227.617 Y-141.467 Z-3.302 I-0.899 J0.731 K0.000 F1143.000
G1 X227.617 Y-141.467 Z-6.604 F762.000
G2 X227.701 Y-141.350 Z-6.604 I0.983 J-0.614 K0.000 F1143.000
G2 X229.499 Y-142.813 Z-6.604 I0.899 J-0.731 K0.000 F1143.000
G2 X227.617 Y-141.467 Z-6.604 I-0.899 J0.731 K0.000 F1143.000
G0 Z7.620
G0 Z7.620
(Finish operation: Z Problem -2 Dowel Holes x 0.332 x 0.390,  0.000 pass ext)

I I select operation  Z Problem -2 Dowel Holes x 0.332 x 0.390,  0.000 pass ext then edit  and then OK (no change made) I get 
[image]
I have no idea what this is telling me.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-11:
Which diameter of the tool do you use for this operation?

tarman3, 2025-01-11:
Changed **Start Depth** and **Finish Step Down** to get this
[image]

Tried, But cant to reproduce on simple body
This bug happens unpredictably

Lin1Jay1, 2025-01-11:
> There are a lot of bugs in CAM Try to use **Helix** instead of **Pocket** ![Screenshot_20250111_174519_lossy](https://private-user-images.githubusercontent.com/188672424/402274075-959991ae-d74c-4241-9d0f-1cbc142a4d43.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY2MjE2NTIsIm5iZiI6MTczNjYyMTM1MiwicGF0aCI6Ii8xODg2NzI0MjQvNDAyMjc0MDc1LTk1OTk5MWFlLWQ3NGMtNDI0MS05ZDBmLTFjYmMxNDJhNGQ0My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMVQxODQ5MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZjY5MzM2MzE5MmQzYWE2YzRmNjQwMWNmZGZmNjBiOGU3YzdiYmZjYzM5MGVmZDk4MWM5YTkwN2FkYmVkNTQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.npc5C9e2w-Chtwm7KFadfVdPGflL7fk8NmhBrTu59t8)
> 
> Which diameter of the tool do you use for this operation?

6mm End Mill

Lin1Jay1, 2025-01-11:
> Decreased **Final Depth** by 0.01 in
> 
> ![Screenshot_20250111_182257_hor](https://private-user-images.githubusercontent.com/188672424/402276857-9268f860-c705-4e43-bcbb-f65e8918b701.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY2MjE2NTIsIm5iZiI6MTczNjYyMTM1MiwicGF0aCI6Ii8xODg2NzI0MjQvNDAyMjc2ODU3LTkyNjhmODYwLWM3MDUtNGU0My1iY2JiLWY2NWU4OTE4YjcwMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMVQxODQ5MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNGZhZmJjNDFmODdkYmQxNmVkZjdkMjVhN2Y2ZGViMGQ4OGQwMTkxMDY0MmQwN2MzOTE1MTk4MmIyNDc3NWE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.WN3659NDvMTKYL8enE1lKpcICtNabUqVZs98L5abY9o)
> 
> Tried, But cant to reproduce on simple body This bug happens unpredictably

That's good to know.  I've been struggling with the CAM part and thinking it must be something I wasn't understanding.  But overtime I've gotten the impression it's early days for CAM.  Reminds me of SWX95 - a memorable experience!!!!

tarman3, 2025-01-11:
Better than nothing )

Lin1Jay1, 2025-01-12:
> @Lin1Jay1 this looks like issue #17748
> 
> Your attached file works for me if I open in RC2 and recompute the operation, but not in 1.0.0

Thanks good to know.  I had some real problems with v1.0.0 and switched back to RC2.  Then I had a problem there and tried V1.0.0.  All my problems seem to be CAM related.

Lin1Jay1, 2025-01-12:
> @Lin1Jay1 this looks like issue #17748
> 
> Your attached file works for me if I open in RC2 and recompute the operation, but not in 1.0.0

Lin1Jay1, 2025-01-12:
> Better than nothing )

Yes it is.  I'm pretty happy all things considered.  Just knowing it's early days is an important reference point for me.  Now when I am having problems I will report as an issue - until someone complains and I get another reference point.  I think I should he able to be useful / helpful.
This is not my 1st experience with gcode.  
[image]
I made this 'Esthetically Pleasing Block' in 1971 which was early days for CAM.  4th year Mech Eng. project.  Wrote Fortran programs to generate gcode to clean out rectangular area, ellipse, roughing and then generate vane.  These programs generated computer cards and those were fed into another machine to generate a tape.  Tape then put into 2.5D machine and you hoped your program worked.  Biggest problem was machine was new and there was no coolant.  But that was 48 years ago.

tarman3, 2025-01-20:
Add parameters BooleanFuzzy
Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean

New float item
BooleanFuzzy = 0.1
Save to disk

Close FreeCAD
Run FreeCAD
Open and close Pocket operation or apply Recompute

After that I got correct depth

[image]

Lin1Jay1, 2025-01-20:
> https://forum.freecad.org/viewtopic.php?t=94034
> 
> Try to add parameters **BooleanFuzzy** Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean
> 
> New float item BooleanFuzzy = 0.1 Save to disk
> 
> [image]
> 
> Close FreeCAD Run FreeCAD Open and close Pocket operation or apply Recompute
> 
> After that I got correct depth
> 
> [image]

I don't get BooleanFuzzy sub directory

[image]

Lin1Jay1, 2025-01-20:
> I don't get BooleanFuzzy sub directory
>
> [image]

Using   OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9

tarman3, 2025-01-20:
You must create this new item
Press right mouse button

[image]

tarman3, 2025-01-20:
Did it work?

Lin1Jay1, 2025-01-20:
> Did it work?

[image]
Then OK

[image]
so no, it did not seem to work.

tarman3, 2025-01-20:
More carefully))

https://github.com/user-attachments/assets/4c92dc4c-0b93-4b93-a600-5c7f9ed5cffc

Add parameters **BooleanFuzzy**
Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean

New float item
BooleanFuzzy = 0.1
Save to disk

Lin1Jay1, 2025-01-21:
> More carefully))
> 
>  obs_2025-01-20_21-56-33_2-25.mp4 
> Add parameters **BooleanFuzzy** Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean
> 
> New float item BooleanFuzzy = 0.1 Save to disk
> 
> Close FreeCAD Run FreeCAD Open and close Pocket operation or apply Recompute

Yes that seemed to work.  I had tried new float item but didn't understand that I had to put in BooleanFuzzy as a name and then get an opportunity to enter a value.  Haven't had to do this kind of thing since AutoCad and DOS.
Appreciate your help.  I will check pockets and let you know.

Lin1Jay1, 2025-01-21:
> More carefully))
> 
>  obs_2025-01-20_21-56-33_2-25.mp4 
> Add parameters **BooleanFuzzy** Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean
> 
> New float item BooleanFuzzy = 0.1 Save to disk
> 
> Close FreeCAD Run FreeCAD Open and close Pocket operation or apply Recompute

Created a simple hole model. and got 

[image]
I don't understand why FC is showing there's a proble.
But CAM generated correct gcode for a couple of step downs.
Thanks for your help.
One of the things that's driving me crazy is when you are in inches and you enter depths in CAM you always have to enter inches.  Is there someplace where you can request improvements or does everything have to go thru the Issue process.
Another example is when you ask to see gcode - just to have a look at it and then cancel it still goes into the save mode.

tarman3, 2025-01-21:
CAM every time show some errors
I stopped paying attention to it

Lin1Jay1, 2025-01-21:
Good to know.  I ignore them because I have no clue what they are telling me.  Sometimes I hit a dead end using v1.0.0 and switch to RC02.  Yesterday V1.0.0 would not produce gcode - only error messages.  So opened in RC02 and gcode was no problem.
Just had a case working on a part for the trays I am making.  I have already made this part and it fit my test tray perfectly.  For my next tray it needs to be 1/16" thicker.  Before doing the machining I have been checking FreeCAD just to be sure there are no surprises.  Two of the profiles that needed to be Outside were now Inside and had to be fixed.   There's no reason for this.  I find that often FreeCAD loses the base geometry and it has to be restored.  But in both cases the base geometry was there.  The only thing that was changed was the depth of machining and step size for the final outer profile with tags.  And tags are another issue.  Had to restore tags that were there but not checked off anymore.

Lin1Jay1, 2025-01-22:
V1.0.0
File: 12x18x0.94-InlayFrame-MachBot-1.22.25-2.FCStd
When programmed direction was Conventional and that’s typical for Profile.  Conventional on an inside path is CCW.  
Now on editing, V1.0.0 is showing it as CW.  The options for Profile are supposed to be Conventional or Climb

[image]

Simulation is CW.
I discovered this problem when I ran the gcode generated by V1.0.0.  The 1st 2 profiles ran CW and the last ran CCW.  So this seems like another case of V1.0.0 not generating correct gcode.

This is an inside path and needs to be CCW to reduce wood splintering.  
If I try to change it to CCW in V1.0.0 I get error messages:
  
[image]

Opened up same file in RC02, and it opens up correctly.

[image]
 
[image]

All 3 profile operations are programmed for Conventional
Simulation: All 3 profiles run CCW.
Gcode generated by RC02 runs CCW and is correct.
So RC02 is working correctly, V1.0.0 is not.

[12x18x0.94-InlayFrame-MachBot-1.22.25-2.FCStd.zip](https://github.com/user-attachments/files/18512518/12x18x0.94-InlayFrame-MachBot-1.22.25-2.FCStd.zip)

tarman3, 2025-01-23:
Looks the problem you described is relative to situation than you created **Job** in FreeCAD  RC02, but after opened it in FreeCAD 1.0

In your file with FreeCAD 1.0 I created new **Job** and there no problem to switch CW/CCW 

https://github.com/user-attachments/assets/05969d8a-f43c-4517-b6b5-49e31aa0d0a4

tarman3, 2025-01-23:
Check this and if i`m right
try to create your next new project from beginning in FreeCAD 1.0

Lin1Jay1, 2025-01-23:
I've been working on this project for about 2-3 months in the process of learning FreeCAD for the 1st time so **initial** model was probably created in RC02.  I would guess initial CAM was done in V1.0.0 but can't be absolutely certain of that.  But by the time I was done proving out the CAM and I had completed my 1st test tray the file was a real mess and generating lots of errors messages.  I started a new file and deleted all the CAM but left the RC02 model in place.
I then re-programmed the CAM in V1.0.0 to clean everything up.  I can't guarantee that it was done in RC02 but it's logical that I would do it this way. - use the latest version  There were times, during optimization of the CAM that V1.0.0 would refuse to generate gcode and would start producing error messages instead.  When that happened, I'd close the file and try to reopen it in V1.0.0.  If that failed, and it generally did, I'd open it in RC02 which would generate gcode required without a problem.  Once in RC02 I probably stayed there until I had a problem and had to switch to V1.0.0 to solve it.
I just added a profile to a simple test model in V1.0.0 to see what profile looks like.

[Pocket Test.zip](https://github.com/user-attachments/files/18523921/Pocket.Test.zip)
Options are now CW and CCW which is probably more understandable.  Virtual simulation is CCW as programmed.

Lin1Jay1, 2025-01-23:
Created a simple surfacing program for ½” bit because my 1.14 dia. bit was burning the wood a bit.

[Surface 19 x 12.875 x 1.12 Start at 0 to -.06-0.5Dia EM.FCStd.zip](https://github.com/user-attachments/files/18527394/Surface.19.x.12.875.x.1.12.Start.at.0.to.-.06-0.5Dia.EM.FCStd.zip)
Had to fudge the tool because I can't create new tools.
I added the 1/2" bit facing operation and FC will not generate gcode.  But I can tell I still have a depth problem because it's showing only 2 passes and what I've asked for requires 3 passes.
Simulation does 2 passes.
When I ask for gcode in V1.0.0 I get 

[image]
In RC02 same thing 

[image]
Even RC02 won't generate gcode

tarman3, 2025-01-24:
Strange
I opened your file and did not see this error

Can you login in forum.freecad.org ???

Lin1Jay1, 2025-01-24:
[image]
still getting error in v1.0.0
Are you telling me you and create gcode and it does 3 passes?
Not sure where Forum is but I'll look

tarman3, 2025-01-24:
Changed **BooleanFuzzy** to 0.001 to get correct depth of **MillFace**

https://github.com/user-attachments/assets/45d15434-ebc6-4cb7-982b-bb7130d35116

We turned it into a chat, which is not good
Please get access to the https://forum.freecad.org

Lin1Jay1, 2025-01-24:
Can’t seem to login.   I’m in as Lin1Jay1 as far as I know and I know my password.  I’ve tried the only 2 emails I use and FreeCAD or github has not sent any emails.  

 

https://github.com/user-attachments/assets/45d15434-ebc6-4cb7-982b-bb7130d35116

tarman3, 2025-01-24:
Wrote  request to one of the admin in forum
Hope he will help

Lin1Jay1, 2025-01-24:
Changed value, still have a problem with pocket – only 2 passes.  Gcode 

Then I tried Mill Face and it is producing 3 passes – but so is the pocket now!

At first pocket was 2 passes and I have the gcode to prove it.  Then I added Mill Face and magically the pocket was fixed.  

Links sent by admin?:  One opened to this conversation.  One was just a black screen.

If you want, you can contact me at ***@***.*** ***@***.***>  then I’ll send you my regular email and delete this.

I’m going to surface a tray.

Jay

 

https://github.com/user-attachments/assets/45d15434-ebc6-4cb7-982b-bb7130d35116

Lin1Jay1, 2025-01-24:
Lin1Jay1    at  gmail.com

tarman3, 2025-01-25:
> Changed value, still have a problem with pocket – only 2 passes. Gcode Then I tried Mill Face and it is producing 3 passes – but so is the pocket now!

Hope problem was solved

Lin1Jay1, 2025-01-25:
Yes basic depth problem seemed fixed for now.
My real problem is I don't have access to tool crib.  I did the add on but have not been able to access it..
I'm just concentrating on working in my shop and optimizing tray CAM.

tarman3, 2025-01-25:
> My real problem is I don't have access to tool crib. I did the add on but have not been able to access it..

Sorry I dont understand

Lin1Jay1, 2025-01-26:
It's a separate issue I guess. 
https://github.com/FreeCAD/FreeCAD/issues/18555#issuecomment-2558596878
Someone suggested add on of Better Tool Library.  I did and it's not available.
When I go to Tool Bit Doc I get

[image] 
and nothing works.  Area where you should have access to libraries has never worked.  Always says Default and there's nothing there.    if I click on edit pencil it flickers and hemorrhages but nothing changes .
It opens a a tool file for a 6,, EM i use regularly and produces an error message.
So I can't create tools using FC.
I've ignored the issue  basically because I'm trying to get something done.  I just needed a 1/2 end mill and managed to use a test bit I had already on file and modified it to meet my need.

tarman3, 2025-01-26:
[image]

Lin1Jay1, 2025-01-26:
Not sure what Icon that is.  Create tool maybe.
When I try to customize CAM Tool bar it's empty.  as are all other Tool bars.
I did create a custom tool bar that might do the trick.

[image]

Ultimately I think I'm just going to have to uninstall everything and start over.
Problem seemed to start when FC asked me where I wanted to keep my tools and then it couldn't keep track of that.

tarman3, 2025-01-26:
There are two ways to get modern tool menu

https://github.com/user-attachments/assets/39de6778-46bc-46e7-b5be-79c6d876e92a

tarman3, 2025-01-26:
You can add icon to custom toolbar

https://github.com/user-attachments/assets/7a4f4fe2-33a4-439c-988d-c86096179670

tarman3, 2025-01-27:
> Can’t seem to login. I’m in as Lin1Jay1 as far as I know and I know my password. I’ve tried the only 2 emails I use and FreeCAD or github has not sent any emails.

Forums admin can find user with login name **Lin1Jay1**
Probably you used another nick name for registration

Lin1Jay1, 2025-01-27:
If admin  ‘can find’ Lin1Jay1 why is there a problem?

tarman3, 2025-01-27:
> If admin ‘can find’ Lin1Jay1 why is there a problem?

I did mistake and changed my message
**Can NOT find**

FEA-eng, 2025-01-27:
@Lin1Jay1 I'm one of the forum admins. Please write to _admin at freecadforum.org_ (of course, you have to replace _at_ with @) so that we can help you directly.

Lin1Jay1, 2025-01-27:
I did not know about the create tool option.  So thank you, that’s very helpful.  As I indicated the tool selector just does nothing now.  It used to work and I used it as indicated until it stopped working.  I have added the Better Tool Library – says it’s installed – but I have no idea where it is.

FreeCAD has access to ~something like User/roaming ?? path but I can’t access it / find it even as administrator.  

 

There are two ways to get new tool menu

https://github.com/user-attachments/assets/39de6778-46bc-46e7-b5be-79c6d876e92a

Lin1Jay1, 2025-01-27:
Yes I have added a custom tool bar and that helps.  Don’t understand why I can’t modify an existing toolbar.

 

https://github.com/user-attachments/assets/7a4f4fe2-33a4-439c-988d-c86096179670

Lin1Jay1, 2025-01-27:
I’m talking to you on   https://github.com/FreeCAD/FreeCAD/issues/18994#issuecomment-2614535768   as Lin1Jay1 and FreeCAD / github has no idea who I am.  Interesting.

tarman3, 2025-01-27:
Please, Try to register in forum again
Take a screenshot when you will write login name and email address

Lin1Jay1, 2025-01-27:
I am quite sure I typed Lin1Jay1 but it came up LinJay and had a password saved.  Tried that then tried typing in my password

Then 

Then 

I haven’t tried contacting Admin

tarman3, 2025-01-27:
[image]

tarman3, 2025-01-27:
I am always happy to help, but it’s extremely not convenient for me to do it on GitHub
I really hope that you will solve the issue with access to the forum

tarman3, 2025-01-27:
Ok
I will write them to activate your account manually

tarman3, 2025-01-27:
Done
Waiting )

tarman3, 2025-01-28:
Please try to login in forum
Take screen shot if it is still not possible

Lin1Jay1, 2025-01-28:
I have logged into Forum but no further ahead.

[image]

tarman3, 2025-01-28:
Do you see any topic?

[image]

Lin1Jay1, 2025-01-28:
Yes I see all kinds of topics

tarman3, 2025-01-28:
Great
Probably this two topics is most useful for you

https://forum.freecad.org/viewforum.php?f=15
https://forum.freecad.org/viewforum.php?f=3

Write your new questions there

tarman3, 2025-01-28:
This issue finally can be closed

closed 2025-01-28 by Lin1Jay1