FreeCAD/FreeCAD #18995: CAM: Creation of 3D Surface operation hangs

bgamari, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to create a 3D Surface operation on the surface of a part results in FreeCAD entering an infinite loop:

1. Open the attached file.
2. Enter the CAM workbench
3. Select any of the faces on the +Z side of `Job001`'s  model
4. Click on the "3D Surface" button
5. Observe that FreeCAD hangs indefinitely

The backtrace while hung will look something like:

```
(gdb) bt
#0  0x00007f8f1c75ebd6 in boost::geometry::strategy::disjoint::detail::box_box<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt>, 0ul, 3ul>::apply (box1=..., box2=...) at include/boost/geometry/strategies/cartesian/disjoint_box_box.hpp:50
#1  0x00007f8f1c75ece4 in boost::geometry::strategy::disjoint::cartesian_box_box::apply<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt> > (box2=..., box1=...) at include/boost/geometry/strategies/cartesian/disjoint_box_box.hpp:83
#2  boost::geometry::dispatch::disjoint<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt>, 3ul, boost::geometry::box_tag, boost::geometry::box_tag, false>::apply<boost::geometry::strategies::relate::cartesian<void> > (strategy=..., box2=..., box1=...)
    at include/boost/geometry/algorithms/detail/disjoint/box_box.hpp:81
#3  boost::geometry::resolve_strategy::disjoint<boost::geometry::default_strategy, false>::apply<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt> > (geometry2=..., geometry1=...) at include/boost/geometry/algorithms/detail/disjoint/interface.hpp:97
#4  boost::geometry::resolve_dynamic::disjoint<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt>, false, false>::apply<boost::geometry::default_strategy> (strategy=..., geometry2=..., geometry1=...)
    at include/boost/geometry/algorithms/detail/disjoint/interface.hpp:129
#5  boost::geometry::disjoint<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt> > (geometry2=..., geometry1=...) at include/boost/geometry/algorithms/detail/disjoint/interface.hpp:231
#6  boost::geometry::intersects<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt> > (geometry2=..., geometry1=...) at include/boost/geometry/algorithms/detail/intersects/interface.hpp:108
#7  boost::geometry::index::detail::spatial_predicate_intersects<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt>, boost::geometry::box_tag, boost::geometry::box_tag>::apply<boost::geometry::default_strategy> (g2=..., g1=...)
    at include/boost/geometry/index/detail/predicates.hpp:218
#8  boost::geometry::index::detail::spatial_predicate_call<boost::geometry::index::detail::predicates::intersects_tag>::apply<boost::geometry::model::box<gp_Pnt>, boost::geometry::model::box<gp_Pnt>, boost::geometry::default_strategy> (s=..., g2=..., g1=...)
    at include/boost/geometry/index/detail/predicates.hpp:243
#9  boost::geometry::index::detail::predicate_check<boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false>, boost::geometry::index::detail::value_tag>::apply<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::model::box<gp_Pnt>, boost::geometry::default_strategy> (s=..., i=..., p=...) at include/boost/geometry/index/detail/predicates.hpp:288
#10 boost::geometry::index::detail::predicates_check_impl<boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false>, boost::geometry::index::detail::value_tag, 0ul, 1ul>::apply<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::model::box<gp_Pnt>, boost::geometry::default_strategy> (s=..., i=..., v=..., p=...) at include/boost/geometry/index/detail/predicates.hpp:599
#11 boost::geometry::index::detail::predicates_check<boost::geometry::index::detail::value_tag, boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false>, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::model::box<gp_Pnt>, boost::geometry::default_strategy> (s=..., i=..., v=..., p=...) at include/boost/geometry/index/detail/predicates.hpp:631
#12 boost::geometry::index::detail::rtree::visitors::spatial_query_incremental<boost::geometry::index::rtree<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, WireJoiner::BoxGetter, boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> >, boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> > >::members_holder, boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false> >::search_value (this=0x4b4b0da8) at include/boost/geometry/index/detail/rtree/visitors/spatial_query.hpp:216
#13 0x00007f8f1c75ef08 in boost::geometry::index::detail::rtree::visitors::spatial_query_incremental<boost::geometry::index::rtree<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, WireJoiner::BoxGetter, boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> >, boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> > >::members_holder, boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false> >::increment (this=<optimized out>) at include/boost/geometry/index/detail/rtree/visitors/spatial_query.hpp:172
#14 boost::geometry::index::detail::rtree::iterators::spatial_query_iterator<boost::geometry::index::rtree<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, WireJoiner::BoxGetter, boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> >, boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> > >::members_holder, boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false> >::operator++ (this=<optimized out>) at include/boost/geometry/index/detail/rtree/query_iterators.hpp:103
#15 boost::geometry::index::detail::rtree::iterators::query_iterator_wrapper<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::detail::rtree::allocators<boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> >, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, boost::geometry::model::box<boost::geometry::model::point<double, 3ul, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::node_variant_static_tag>, boost::geometry::index::detail::rtree::iterators::spatial_query_iterator<boost::geometry::index::rtree<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, WireJoiner::BoxGetter, boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> >, boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> > >::members_holder, boost::geometry::index::detail::predicates::spatial_predicate<boost::geometry::model::box<gp_Pnt>, boost::geometry::index::detail::predicates::intersects_tag, false> > >::increment (this=<optimized out>)
    at include/boost/geometry/index/detail/rtree/query_iterators.hpp:247
#16 0x00007f8f1c76b631 in boost::geometry::index::detail::rtree::iterators::query_iterator<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::detail::rtree::allocators<boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> >, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16ul, 4ul>, boost::geometry::model::box<boost::geometry::model::point<double, 3ul, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::node_variant_static_tag> >::operator++ (this=0x7ffd6e761628)
    at include/boost/geometry/index/detail/rtree/query_iterators.hpp:325
#17 WireJoiner::splitEdges (this=this@entry=0x7ffd6e761d30) at /build/source/src/Mod/CAM/App/Area.cpp:1118
[... cut ...]
OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.Unknown
Build type: Release
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Beltrami 1.1.0
  * OpenTheme 2024.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.52
```

### Subproject(s) affected?

CAM

### Anything else?

Thanks for a great tool!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgamari, 2025-01-11:
`boxMap` is :
```
(gdb) print boxMap
$3 = {
  m_members = {<boost::geometry::index::detail::translator<WireJoiner::BoxGetter, boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> > >> = {<WireJoiner::BoxGetter> = {<No data fields>}, <boost::geometry::index::equal_to<std::_List_iterator<WireJoiner::EdgeInfo> >> = {<boost::geometry::index::detail::equal_to<std::_List_iterator<WireJoiner::EdgeInfo>, false>> = {<No data fields>}, <No data fields>}, <No data fields>}, <boost::geometry::index::linear<16, 4>> = {
      static max_elements = 16}, <boost::geometry::index::detail::rtree::allocators<boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> >, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16, 4>, boost::geometry::model::box<boost::geometry::model::point<double, 3, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::node_variant_static_tag>> = {<boost::container::new_allocator<boost::variant<boost::geometry::index::detail::rtree::variant_leaf<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16, 4>, boost::geometry::model::box<boost::geometry::model::point<double, 3, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::allocators<boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> >, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16, 4>, boost::geometry::model::box<boost::geometry::model::point<double, 3, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::node_variant_static_tag>, boost::geometry::index::detail::rtree::node_variant_static_tag>, boost::geometry::index::detail::rtree::variant_internal_node<std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16, 4>, boost::geometry::model::box<boost::geometry::model::point<double, 3, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::allocators<boost::container::new_allocator<std::_List_iterator<WireJoiner::EdgeInfo> >, std::_List_iterator<WireJoiner::EdgeInfo>, boost::geometry::index::linear<16, 4>, boost::geometry::model::box<boost::geometry::model::point<double, 3, boost::geometry::cs::cartesian> >, boost::geometry::index::detail::rtree::node_variant_static_tag>, boost::geometry::index::detail::rtree::node_variant_static_tag> > >> = {<No data fields>}, <No data fields>}, 
    values_count = 13442, leafs_level = 4, root = 0x216c5a00}}
(gdb) 
```

maxwxyz, 2025-01-11:
@bgamari does it work in the stable 1.0 version?

bgamari, 2025-01-12:
> @bgamari does it work in the stable 1.0 version?

FreeCAD 1.0 with the same `opencamlib` version fails with the same failure mode.

sliptonic, 2026-04-23:
Sounds like there's nothing actionable.  Closing.