FreeCAD/FreeCAD #18996: Measuring tool labels don't scale to set text size

drejc, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have changed the Measure -> Appearance -> Text size to 30px. The main measuring label is scaled appropriately while the rest is not.

[image]

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

baldhead69, 2025-04-28:
Same problem here

luzpaz, 2025-05-10:
@hlorus any chance you can have a look at this and weigh-in ?

PR #21467 (2025-05-20): Measure: Fix delta label text size scaling
This solves #18996 by correctly hooking up the text size properties from ViewProviderMeasureBase->DimensionLinear->SoFrameLabel.

- Add font size field to ViewProviderMeasureBase which connects to other fields
- Remove dead code in DimensionLinear which internally is using FrameLabel
- Connect FrameLabels's fontsize property to DimensionLinear's property

## Issues

fixes #18996

## Before and After Images

Before:
[image]

After:
![image](https://github.com/user-attachments/assets/3257330a-8c70-4974-8f43-11b80a9d825
closed 2025-05-20 by hyarion