FreeCAD/FreeCAD #18998: part design selcting edge highlights entire solid

JaapStruyk, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Entire solid is highlighted when selecting an edge in 1.1.x and 1.0 
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-11:
Please add an example file and steps how to reproduce

JaapStruyk, 2025-01-11:
Problem excist only with object placed in a link group `Link actions:Make link group:Simple group`
[image]
[image]
[Unnamed1.zip](https://github.com/user-attachments/files/18387174/Unnamed1.zip)