FreeCAD/FreeCAD #18999: part design sketch line color as set in tree view not saved

JaapStruyk, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Wgen selecting a different object line color for a sketch in part design after reopenening drawing color is back to default, both 1.1.dev and 1.0
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-11:
could you please add an example file?
@kadet1090 iirc the auto color should only change depending on the theme if the color has not been changed?

JaapStruyk, 2025-01-11:
[image]
[image]
[walsrol.zip](https://github.com/user-attachments/files/18387149/walsrol.zip)

kadet1090, 2025-01-12:
> could you please add an example file? @kadet1090 iirc the auto color should only change depending on the theme if the color has not been changed?

No, color changes automatically if `Auto Color` property is set to true, which is the case in the screenshot. Not sure why color is even visible in that case as it should be hidden. Nonetheless, @JaapStruyk set `Auto Color` to false and then change the color, it should persist.

maxwxyz, 2025-01-12:
Should this issue then left open for showing the ability to show color input despite being autocolor?

kadet1090, 2025-01-12:
If we are able to provide reproduction steps - yes.

maxwxyz, 2025-01-12:
@JaapStruyk can you provide the steps you have done to reproduce this?

JaapStruyk, 2025-01-13:
Setting `auto color:false` indeed works so no bug!
Maybe hide color settings when autocolor is set to true or remove autocolor option and default it to yes unless a color is selected.

kadet1090, 2025-01-13:
> Setting `auto color:false` indeed works so no bug! Maybe hide color settings when autocolor is set to true or remove autocolor option and default it to yes unless a color is selected.

It should work like that. If auto color is set to true exact color should be marked as hidden. Not sure why it does not do that.

luzpaz, 2025-05-11:
Any body able to propose a fix ?

Roy-043, 2025-11-11:
@JaapStruyk can you test with the latest dev version? I can reproduce that toggling AutoColor does not (un)hide color and line width fields in v1.0.2, but AFAICT this problem does not occur in the current dev version.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

JaapStruyk, 2025-11-12:
Last devs working as it should, autocolor yes shows no color changing options, set to no and changing color of sketches it's still there after saving, issue resolved :-)