FreeCAD/FreeCAD #19001: [macOS] QuickLook with new API since macOS 15 (Sequoia)

ryanthara, 2025-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The QuickLook functionality introduced to FreeCAD with [this](https://github.com/FreeCAD/FreeCAD/pull/7491) pull request works only until macOS 10.14. With the macOS Sequoia release a new API has to be used.

`Support for deprecated Quick Look Generator plugins is being removed. To provide previews and thumbnails for your custom file types, migrate to Quick Look Preview Extension and Thumbnail Extension API. (116791365)`

See https://developer.apple.com/documentation/macos-release-notes/macos-15-release-notes

### Full version info

```shell
OS: macOS 15.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Trails 2022.1.0
  * ThreadProfile 1.81.0
  * Reporting
  * Behave-Dark-Colors 0.0.1
  * dodo
  * DynamicData 2.70.0
  * parts_library
  * fcgear 1.0.0
  * workfeature
  * fasteners 0.3.40
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * Assembly4 0.11.10
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-11:
will the new API break it on macOS 13?

luzpaz, 2025-01-11:
cc @MatthiasWM any chance you could weigh-in here ?

MatthiasWM, 2025-01-12:
Thanks for the update. I will try to find some time to look at this tonight and reply here.

hyarion, 2025-03-02:
> Thanks for the update. I will try to find some time to look at this tonight and reply here.

Any updates on this @MatthiasWM? :)

MatthiasWM, 2025-05-08:
Sorry, real life got in the way. I will try to give this another shot this weekend, but no promises.

luzpaz, 2025-05-08:
No pressure, thanks for your efforts.

graelo, 2025-07-19:
Hi, I've just pushed a new preview app extension for macOS 15+: https://github.com/graelo/FreeCAD-preview-extension. I did this for me but hoping this could be integrated in the FreeCAD bundle.

It's simple: using Swift and the new API, the extension simply opens the FCStd file and displays the Thumbnail.png. 

I've been using it locally for a couple months.

I was looking at the previous PR on the topic 3 years ago:
- https://github.com/FreeCAD/FreeCAD-Bundle/pull/136
- https://github.com/FreeCAD/FreeCAD/pull/7491
but the integration work seems too complex for my limited understanding of this part. Can someone help for this part? @luzpaz @MatthiasWM maybe (sorry for tagging)

PS: I set the license to MIT, I hope this is not an issue. I can still change it.
PS2 : I set an icon, derived from the original logo (just added a looking glass), this can be dropped.

Thanks!

hyarion, 2025-07-19:
@adrianinsaval @oursland any ideas?

oursland, 2025-07-20:
The location of the current QuickLook plugin is here: https://github.com/FreeCAD/FreeCAD/tree/main/src/MacAppBundle/QuickLook

This is installed into the `Contents/Library/QuickLook` path by this `CMakeLists.txt`: https://github.com/FreeCAD/FreeCAD/blob/main/src/MacAppBundle/CMakeLists.txt

Is it possible to adapt this to the new QuickLook extension?

TH1813254617, 2025-10-20:
Blender only has one Quicklook extension, like @graelo implementation.
[image]
They use it to produce thumbnails, Quick Look Preview for .blend files is a scaled up (full resolution?) thumbnail.
[image]
Blender's thumbnail seem to to have low resolution, I can't tell if it's the full thumbnail PNG or just the icon zoomed in. If it's the full PNG I don't see a need for two Quick Look extensions -- changing Graelo's implementation to generate only thumbnails would fix both thumbnails and previews.

The only reason to have a seperate Quick Look preview extension seems to be extra functionality, like scrolling through LibreOffice documents and rotating 3D renders. With FreeCAD, the use seems limited as discussed [here](https://forum.freecad.org/viewtopic.php?f=10&t=71803). Unless something changed and we can now have FreeCAD quickly generate a 3D model for use in Quick Look preview.

graelo, 2025-10-21:
Hi, I now have a working thumbnail preview and I kept the full size preview too (it looks better than the scaled thumbnail). I'll push the code onto my repo within a few hours.

Here's how this looks

The new QLThumbnail preview

[image]

and

[image]

And also the existing QuickLook Preview

[image]

graelo, 2025-10-21:
After the code is pushed, I'll investigate how to embed this instead of the existing implementation (thanks @chennes) and report back here. Cheers

TH1813254617, 2025-10-23:
Hi @graelo. The extensions work.

Just one small issue, the minimum deplayment for the Thumbnail Extension is set to MacOS 26 instead of Sequioa 15.5 like the others. I had to change it to make it work for me -- I'm still on Sequioa.

closed 2026-01-21 by chennes

graelo, 2026-04-08:
Hi, this will hopefully be available in 1.2
In the meantime, you can find it in the weekly releases (in the macOS 15 dmg): https://github.com/FreeCAD/FreeCAD/releases
Cheers