FreeCAD/FreeCAD #19008: FreeCAD v1.0 (Linux) only loads and saves FCStd files within the home folder

hmf, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The FCStd files can only be opened and saved in a folder in the "home" folder. In a mount that I have full privileges loading of saving a file fails. 

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Core

### Anything else?

This works:

```shell
:~/dev/cad$ freecad test3.FCStd
```

This does not:

```shell
:/mnt/ssd2/hmf/VSCodeProjects/smartphones/docs/cad$ freecad test3.FCStd 
FreeCAD 1.0.0, Libs: 1.0.0R38641 +468 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/hmf/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1248/lib/python3.10/site-packages:/snap/freecad/1248/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
<Exception> Invalid project file: test3.FCStd
Exception while processing file: test3.FCStd [Invalid project file: test3.FCStd]
```

Here is the report view:

```shell
11:08:30  <Exception> Invalid project file: test3.FCStd
11:08:30  Exception while processing file: test3.FCStd [Invalid project file: test3.FCStd]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-12:
Possible duplicate of https://github.com/FreeCAD/FreeCAD-snap/issues/58

hmf, 2025-01-13:
@furgo16 Please mote that this is not removable media. It is a hard disk (SSD) mounted on `/mnt`.

furgo16, 2025-01-13:
> removable-media allows read/write access to mounted removable storage in `/media`, `/run/media` and `/mnt`.

https://snapcraft.io/docs/removable-media-interface