FreeCAD/FreeCAD #19012: Part::Extrude of SubShapeBinder has different normal vector than underlying sketch

j-hap, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a single sketch with multiple profiles to extrude. These simple steps reproduce the issue I encountered:

- Create a sketch with a rectangle on the xy plane
- Create a subshape binder of the whole sketch in Part Design (that is not part of a body)
- Create a Part Extrude of the sketch
- Create a Part Extrude of the subshape binder

Up to here, both extrudes are the same, but notice how the extrude based on the subshape binder switches its extrusion direction when the sketch is rotated more than 45° around the x-axis. this is, of course, also the case for a sketch created on the xz plane.

`Base::Vector3d Extrusion::calculateShapeNormal(const App::PropertyLink& shapeLink)` does special handling when the base is a 2d object, which a subshape binder is not necessarily, therefore the normal calculation is different.

I know I can use wires of a sketch for extrusion in PartDesign and omit the subshape binder, but I want to work with the Part Workbench.

My workaround is to set the extrusion direction to the sketch normal instead of auto calculation, so the issue is avoidable for me, but nevertheless i wanted to report it.

### Full version info

```shell
OS: Fedora Linux 41 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39793 (Git) Conda
Build type: Release
Branch: main
Hash: c42031690377122a80ff574ef596b2a6cec5adbf
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
Could you add a .zipped example file please?

j-hap, 2025-01-19:
here's a file made with the steps described above, reduce the sketch placement angle to see a switch in the extrusion direction of the extrude based on the shape binder
[extrusion_direction_issue.zip](https://github.com/user-attachments/files/18468934/extrusion_direction_issue.zip)

fivethreeo, 2025-09-04:
Is the shapebinded sketch inside a body? I put the sketch in a body. Then it suddenly played nice.

FEA-eng, 2025-11-03:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251102 (Git shallow)
Build date: 2025/11/02 16:03:37
Build type: Release
Branch: main
Hash: 45ca23d074f39a9b586c8320f1e7cd20ca466fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```